sm0svx/svxlink #488: The squelch is opening, but not closing

RedCatN1, 2020-04-09:
I use GPIO pins to control the noise canceler and there was such a problem, the noise canceler opens but does not close, it closes only after the SIGTERM signal when closing the application, this is what is in the terminal:
`root@orangepizero:~# svxlink
SvxLink v1.7.0 Copyright (C) 2003-2019 Tobias Blomberg / SM0SVX

SvxLink comes with ABSOLUTELY NO WARRANTY. This is free software, and you are
welcome to redistribute it in accordance with the terms and conditions in the
GNU GPL (General Public License) version 2 or later.

Using configuration file: /etc/svxlink/svxlink.conf
--- Using sample rate 48000Hz

Starting logic: SimplexLogic
Loading RX: Rx1
Loading TX: Tx1
Loading module "ModuleHelp" into logic "SimplexLogic"
        Found /usr/lib/arm-linux-gnueabihf/svxlink/ModuleHelp.so
        Module Help v1.0.0 starting...
Loading module "ModuleParrot" into logic "SimplexLogic"
        Found /usr/lib/arm-linux-gnueabihf/svxlink/ModuleParrot.so
        Module Parrot v1.1.1 starting...
Loading module "ModuleFrn" into logic "SimplexLogic"
        Found /usr/lib/arm-linux-gnueabihf/svxlink/ModuleFrn.so
        Module Frn v1.1.0 starting...
Loading module "ModuleTclVoiceMail" into logic "SimplexLogic"
        Found /usr/lib/arm-linux-gnueabihf/svxlink/ModuleTcl.so
        Module Tcl v1.0.1 starting...
SimplexLogic: Event handler script successfully loaded.
Rx1: The squelch is OPEN (0)

SIGINT received. Shutting down application...
Rx1: The squelch is CLOSED (0)`
my operating system Armbian buster with Linux 5.4.20-sunxi(Based on Debian) and hardware orangepizero  
and here is my configuration
`[Rx1]
TYPE=Local
#RX_ID=?
AUDIO_DEV=alsa:plughw:0
AUDIO_CHANNEL=0
#AUDIO_DEV_KEEP_OPEN=0
SQL_DET=GPIO
SQL_START_DELAY=0
SQL_DELAY=0
SQL_HANGTIME=2000
#SQL_EXTENDED_HANGTIME=1000
#SQL_EXTENDED_HANGTIME_THRESH=15
#SQL_TIMEOUT=600
VOX_FILTER_DEPTH=20
VOX_THRESH=1000
#CTCSS_MODE=2
CTCSS_FQ=136.5
#CTCSS_SNR_OFFSET=0
#CTCSS_OPEN_THRESH=15
#CTCSS_CLOSE_THRESH=9
#CTCSS_BPF_LOW=60
#CTCSS_BPF_HIGH=270
SERIAL_PORT=/dev/ttyS0
SERIAL_PIN=CTS
#SERIAL_SET_PINS=DTR!RTS
#EVDEV_DEVNAME=/dev/input/by-id/usb-SYNIC_SYNIC_Wireless_Audio-event-if03
#EVDEV_OPEN=1,163,1
#EVDEV_CLOSE=1,163,0
GPIO_PATH=/sys/class/gpio
GPIO_SQL_PIN=gpio3
#PTY_PATH=/tmp/rx1_sql
#HID_DEVICE=/dev/hidraw3
#HID_SQL_PIN=VOL_UP
#SIGLEV_DET=TONE
SIGLEV_SLOPE=1
SIGLEV_OFFSET=0
#SIGLEV_BOGUS_THRESH=120
#TONE_SIGLEV_MAP=100,84,60,50,37,32,28,23,19,8
SIGLEV_OPEN_THRESH=5
SIGLEV_CLOSE_THRESH=2
DEEMPHASIS=0
#SQL_TAIL_ELIM=300
#PREAMP=6
PEAK_METER=1
DTMF_DEC_TYPE=INTERNAL
DTMF_MUTING=1
DTMF_HANGTIME=40
DTMF_SERIAL=/dev/ttyS0
#DTMF_PTY=/tmp/rx1_dtmf
#DTMF_MAX_FWD_TWIST=8
#DTMF_MAX_REV_TWIST=4
#1750_MUTING=1
#SEL5_DEC_TYPE=INTERNAL
#SEL5_TYPE=ZVEI1
#FQ=433475000
#MODULATION=FM
#WBRX=WbRx1
#OB_AFSK_ENABLE=0
#OB_AFSK_VOICE_GAIN=6
#IB_AFSK_ENABLE=0`

sm0svx, 2020-04-19:
Are you sure that your GPIO setup is working? If you stop SvxLink and then run the following command, does it flip between 0/1 on squelch close/open?

`watch -n0.1 cat /sys/class/gpio/gpio3/value`

RedCatN1, 2020-04-20:
I understood what was the reason, the armbian assembly itself was problematic, I assembled my assembly of pure debian on the 5.x.x kernel and everything worked as needed. I assume that the armbian assembly has a problem with the GPIO driver. Thanks for the answer, svxlink is the best RoIP system I've seen

sm0svx, 2020-04-20:
I'm glad that you sorted it out and that you find SvxLink useful!

closed 2020-04-20 by sm0svx