sm0svx/svxlink #489: remotetrx while activating svxserver-connection not working

dl7ata, 2020-04-10:
Dear all,

I am running latest v1.7.99.21 and use a link into a svxserver-connection. Everything runs fine.
I compiled another svxlink on an ubuntu machine which is configured as a remotetrx.
While I am not connected into the netlink, remotetrx runs excellent, using vox on that ubuntu-notebook. When I connect the netlink (91#), signal grom remotetrx is stuttering and no audio is transmitted any more. After a disconnect (9#), everything is back to normal. It looks like, that the squelch-signal transmitting to svxserver may cause trouble.

sm0svx, 2020-04-19:
Have you looked at "top"? Stuttering most often is caused by too high CPU load. If your system does not have enough CPU power it may not be able to handle audio encoding/decoding for all connections simultaneously.

dl7ata, 2020-04-19:
Yes - it's a i7-machine and stuttering means, that it's switching between RX and TX. CPU load stays on the same level. On the same machine, a remotetrx was running quite some time ago (a year or more) with the same system. It seems to be a matter with the squelch, just my imporession.

sm0svx, 2020-04-19:
ok I see what you mean but now I'm even more confused :-/

I have no idea what that could be. Can you say something more about your configuration? Is it just a SimplexLogic with a RemoteTrx and then the ReflectorLogic connected to the SimplexLogic? Any more "components" involved?

dl7ata, 2020-04-20:
Setup is SimplexLogic, RemoteLogic and ReflectorLogic. No more components.

sm0svx, 2020-04-20:
What is the RemoteLogic and how does it relate to the other logics?

dl7ata, 2020-04-20:
RemoteLogic is a 2m TRX, SimplexLogic is a 70cm TRX. So its working as a Crossband-repeater. Both are Motorola GM-340 radios. Squelch is coming from radios, feeded via opto into a bananPi.

dl7ata, 2020-05-11:
I got the bug:
The default remotetrx.con contains **#MUTE_TX_ON_RX=1000**. The mainpage says, that the default value is 0, but it is not. So with **MUTE_TX_ON_RX=0** everything works fine. I'll close this issue. Maybe you can change the sourcecode or set **MUTE_TX_ON_RX=0** into the default conf.

closed 2020-05-11 by dl7ata

sm0svx, 2020-07-25:
@dl7ata, where did you find that default value?  The [current manual](https://github.com/sm0svx/svxlink/blob/master/src/doc/man/remotetrx.conf.5#L175-L180) page seem to be correct, _"The default is not to mute the transmitter when the squelch is open"_, so you must have found some old documentation somewhere.

dl7ata, 2020-07-25:
In the sourcecode of ../src/svxlink/remotetrx/remotetrx.conf I found #MUTE_TX_ON_RX=1000 (commented out). So that means that remotetrx use the default value. It should be not 0, because it doesnt work. I have to use explicity MUTE_TX_ON_RX=0 in the remotetrx.conf, then remotetrx use "0" and everything works fine.
I didnt checked the C++ sourcecode where you define and initialize MUTE_TX_ON_RX.

sm0svx, 2020-07-26:
What you looked at is the template configuration file. The commented out values often does not represent the default values, as implemented in the source code, but rather are example values if you want to use the feature.

The source code and the documentation is in compliance with each other which is the most important thing.

dl7ata, 2020-07-27:
You are right - and I agree with compliance.
But the default installation of remotetrx doesn't work on my computer, until I put MUTE_TX_ON_RX=0 into the config file.