sm0svx/svxlink #490: Linkmanager, some commented out lines

dl1hrc, 2020-04-16:
We have some commented out lines In LinkManager::cmdReceived:
...
 else //if (subcmd == "1") // Connecting Logic1 <---> Logic2 (two ways)                                                                                                          
  {                                                                                                                                                                               
    if (!link.is_activated)                                                                                                                                                       
    {                                                                                                                                                                             
      activateLink(link);                                                                                                                                                         
      ss << "activating_link \"";                                                                                                                                                 
      ss << logic_props.announcement_name + "\"";                                                                                                                                 
    }                                                                                                                                                                             
    //else                                                                                                                                                                        
    //{                                                                                                                                                                           
    //  ss << "link_already_active \"";                                                                                                                                           
    //}                                                                                                                                                                           
    //ss << logic_props.announcement_name + "\"";                                                                                                                                 
    sendCmdToLogics(link, logic, subcmd);                                                                                                                                         
  }                                                                                                                                                                               
  //else                                                                                                                                                                          
  //{                                                                                                                                                                             
  //  ss << "unknown_command " << logic_props.cmd << subcmd;                                                                                                                      
  //}                                                                                                                                                                             
  return ss.str();                                                                                                                                                                
} /* LinkManager::cmdReceived */ 

No idea why it is comment out, if the link is already active the tcl procedure should be called up to announce it.
73s Adi / DL1HRC

sm0svx, 2020-04-19:
That was changed to make it possible to send commands to the logic link while it is active. That is necessary for ReflectorLogic for example so that users can enter commands to control talkgroup selection among other things.

You could possibly emulate the old behaviour in your logic by implementing the LogicBase::remoteCmdReceived function. You also need to be able to fire off TCL events so you need to set up an event handler in your logic. Have a look in ReflectorLogc.cpp how it's done. If it's a standard SimplexLogic or RepeaterLogic this should probably be implemented in Logic.cpp but I haven't looked at it.

dl1hrc, 2020-04-20:
ok, will take a closer look at this soon