sm0svx/svxlink #494: ModuleMetarInfo setting error

ANDPJL, 2020-04-26:
Hello, here is my Metar Info conf about airports:

AIRPORTS=LIMF

AND HERE THE ERRO DURING START UP:
 
Sun Apr 26 07:28:45 2020: Starting logic: SimplexLogic
Sun Apr 26 07:28:45 2020: Loading RX: Rx1
Sun Apr 26 07:28:45 2020: Loading TX: Tx1
Sun Apr 26 07:28:45 2020: Loading module "ModuleHelp" into logic "SimplexLogic"
Sun Apr 26 07:28:45 2020: 	Found /usr/lib/arm-linux-gnueabihf/svxlink/ModuleHelp.so
Sun Apr 26 07:28:45 2020: 	Module Help v1.0.0 starting...
Sun Apr 26 07:28:45 2020: Loading module "ModuleParrot" into logic "SimplexLogic"
Sun Apr 26 07:28:45 2020: 	Found /usr/lib/arm-linux-gnueabihf/svxlink/ModuleParrot.so
Sun Apr 26 07:28:45 2020: 	Module Parrot v1.1.1 starting...
Sun Apr 26 07:28:45 2020: Loading module "MetarInfo" into logic "SimplexLogic"
Sun Apr 26 07:28:45 2020: 	Found /usr/lib/arm-linux-gnueabihf/svxlink/ModuleMetarInfo.so
Sun Apr 26 07:28:45 2020: 	Module MetarInfo v1.2.1.99.2 starting...
Sun Apr 26 07:28:45 2020: *** ERROR: Config variable MetarInfo/AIRPORTS not set or wrong, example: AIRPORTS=EDDP,EDDS,EDDB
Sun Apr 26 07:28:45 2020: *** ERROR: Initialization failed for module MetarInfo in logic SimplexLogic

ANY SUGGESTION?
THANKS

dl1hrc, 2020-04-26:
Can not confirm this. What SvxLink-version do you use?

ANDPJL, 2020-04-26:
Hello, The last oneArch. Andrea BellaAndrea Bella Concept Studio33, Via Segantini10024 Moncalieri (TO)www.andreabella.itinfo@andreabella.it -------- Messaggio originale --------Da: Adi Bier <notifications@github.com>Data: dom 26 apr 2020, 13:43A: sm0svx/svxlink <svxlink@noreply.github.com>Cc: ANDPJL <info@andreabella.it>, Author <author@noreply.github.com>Oggetto: Re: [sm0svx/svxlink] ModuleMetarInfo setting error (#494)
Can not confirm this. What SvxLink-version do you use?

dl1hrc, 2020-04-26:
> Hello, The last oneArch.
??
SvxLink-Version, **not** the Linux distribution. Please provide your ModuleMetarInfo.conf

ANDPJL, 2020-04-26:
Hello, my svxlink version 1..7.99.21 

here is my Module: 

[ModuleMetarInfo]
NAME=MetarInfo
ID=5
TIMEOUT=120
TYPE=XML
#SERVER=tgftp.nws.noaa.gov
SERVER=https://aviationweather.gov
#LINK=data/observations/metar/stations
LINK=/adds/dataserver_current/httpparam?dataSource=metars&requestType=retrieve&$
STARTDEFAULT=LIMF
LONGMESSAGES=1
REMARKS=1
DEBUG=1
# insert ICAO airport shortcuts here. You can
# request the METAR by sending dtmf commands as
# follows:
#
# 1# 2# 3# 4# 5# ...
AIRPORTS=LIMF 

---
Arch. Andrea Bella

 Andrea Bella Concept Studio
 33, Via Segantini
 10024 Moncalieri (TO)

 www.andreabella.it [3]
 info@andreabella.it 

Informativa Privacy - Ai sensi del Regolamento Europeo (EU) 679/2016
(GDPR ), si precisa che le informazioni contenute in questo messaggio
sono riservate e ad uso esclusivo del destinatario. Qualora il messaggio
in parola Le fosse pervenuto per errore, La preghiamo di eliminarlo
senza copiarlo e di non inoltrarlo a terzi, dandocene gentilmente
comunicazione. Grazie.  

All'indirizzo: https://www.andreabella.it [4] troverà pubblicata la
nostra politica sulla privacy per qualsiasi chiarimento può contattare
il titolare del trattamento all'indirizzo di posta elettronica
INFO@ANDREABELLA.IT 

 

Links:
------
[1] https://github.com/sm0svx/svxlink/issues/494#issuecomment-619544401
[2]
https://github.com/
[3] http://www.andreabella.it
[4] https://www.andreabella.it/

dl1hrc, 2020-04-26:
check your LINK param, must be
LINK=/adds/dataserver_current/httpparam?dataSource=metars&requestType=retrieve&format=xml&hoursBeforeNow=3&mostRecent=true&stationString=

Please check twice that there are no control characters or similar in the config file, especially around AIRPORTS.
This here is works without error:
```
[ModuleMetarInfo]
NAME=MetarInfo
ID=5
TIMEOUT=120
TYPE=XML
#SERVER=tgftp.nws.noaa.gov
SERVER=https://aviationweather.gov
#LINK=data/observations/metar/stations
LINK=/adds/dataserver_current/httpparam?dataSource=metars&requestType=retrieve&format=xml&hoursBeforeNow=3&mostRecent=true&stationString=
STARTDEFAULT=LIMF
LONGMESSAGES=1
REMARKS=1
DEBUG=1
# insert ICAO airport shortcuts here. You can
# request the METAR by sending dtmf commands as
# follows:
#
#         1#   2#   3#   4#   5#  ...
AIRPORTS=LIMF
```

ANDPJL, 2020-04-26:
Hello, Unbelievable!!! Copie and pasted into MetarOnfo module and give me the same error... Arch. Andrea BellaAndrea Bella Concept Studio33, Via Segantini10024 Moncalieri (TO)www.andreabella.itinfo@andreabella.it -------- Messaggio originale --------Da: Adi Bier <notifications@github.com>Data: dom 26 apr 2020, 15:23A: sm0svx/svxlink <svxlink@noreply.github.com>Cc: ANDPJL <info@andreabella.it>, Author <author@noreply.github.com>Oggetto: Re: [sm0svx/svxlink] ModuleMetarInfo setting error (#494)
check your LINK param, must be
LINK=/adds/dataserver_current/httpparam?dataSource=metars&requestType=retrieve&format=xml&hoursBeforeNow=3&mostRecent=true&stationString=
Please check twice that there are no control characters or similar in the config file, especially around AIRPORTS.
This here is works without error:
[ModuleMetarInfo]
NAME=MetarInfo
ID=5
TIMEOUT=120
TYPE=XML
#SERVER=tgftp.nws.noaa.gov
SERVER=https://aviationweather.gov
#LINK=data/observations/metar/stations
LINK=/adds/dataserver_current/httpparam?dataSource=metars&requestType=retrieve&format=xml&hoursBeforeNow=3&mostRecent=true&stationString=
STARTDEFAULT=LIMF
LONGMESSAGES=1
REMARKS=1
DEBUG=1
# insert ICAO airport shortcuts here. You can
# request the METAR by sending dtmf commands as
# follows:
#
#         1#   2#   3#   4#   5#  ...
AIRPORTS=LIMF

ANDPJL, 2020-04-26:
Hello! Solved the problem: in svxlink. Conf the name of the Module was not correct so the module were be loaded. Thank you very much for your attention!!! 73!AndreaArch. Andrea BellaAndrea Bella Concept Studio33, Via Segantini10024 Moncalieri (TO)www.andreabella.itinfo@andreabella.it -------- Messaggio originale --------Da: Adi Bier <notifications@github.com>Data: dom 26 apr 2020, 15:23A: sm0svx/svxlink <svxlink@noreply.github.com>Cc: ANDPJL <info@andreabella.it>, Author <author@noreply.github.com>Oggetto: Re: [sm0svx/svxlink] ModuleMetarInfo setting error (#494)
check your LINK param, must be
LINK=/adds/dataserver_current/httpparam?dataSource=metars&requestType=retrieve&format=xml&hoursBeforeNow=3&mostRecent=true&stationString=
Please check twice that there are no control characters or similar in the config file, especially around AIRPORTS.
This here is works without error:
[ModuleMetarInfo]
NAME=MetarInfo
ID=5
TIMEOUT=120
TYPE=XML
#SERVER=tgftp.nws.noaa.gov
SERVER=https://aviationweather.gov
#LINK=data/observations/metar/stations
LINK=/adds/dataserver_current/httpparam?dataSource=metars&requestType=retrieve&format=xml&hoursBeforeNow=3&mostRecent=true&stationString=
STARTDEFAULT=LIMF
LONGMESSAGES=1
REMARKS=1
DEBUG=1
# insert ICAO airport shortcuts here. You can
# request the METAR by sending dtmf commands as
# follows:
#
#         1#   2#   3#   4#   5#  ...
AIRPORTS=LIMF