sm0svx/svxlink #495: USB SDR gives terrible audio

KayosPi, 2020-04-27:
I have a working Svxlink setup, version 17.12.2-4 running on a Raspberry Pi4

I've tried to add an SDR to improve receive range and no matter how I config I get terrible interference on the audio. SDR is working as expected, rtl_sdr, blacklist and udev configured. If I use devcal I can hear nice clear audio (either transmitting or if I use a broadcast FM frequency)

I get the same plugged into the pi or through an Orange Pi Zero as a remote link

CTCSS and DTMF decoding work perfectly on the SDR input

Config file below, using the audio in and out works as expected

`[GLOBAL]
MODULE_PATH=/usr/lib/arm-linux-gnueabihf/svxlink
LOGICS=SimplexLogic
CFG_DIR=/etc/svxlink/svxlink.d
TIMESTAMP_FORMAT="%c"
CARD_SAMPLE_RATE=48000
CARD_CHANNELS=1
#LOCATION_INFO=LocationInfo
#LINKS=LinkToR4

[SimplexLogic]
TYPE=Simplex
RX=Voter
TX=Local_Tx
MODULES=ModuleHelp,ModuleParrot
CALLSIGN=26AB1234
SHORT_IDENT_INTERVAL=10
LONG_IDENT_INTERVAL=60
#IDENT_ONLY_AFTER_TX=4
#EXEC_CMD_ON_SQL_CLOSE=500
EVENT_HANDLER=/usr/share/svxlink/events.tcl
DEFAULT_LANG=en_US
RGR_SOUND_DELAY=0
#REPORT_CTCSS=136.5
#TX_CTCSS=ALWAYS
MACROS=Macros
FX_GAIN_NORMAL=0
FX_GAIN_LOW=-12
#ACTIVATE_MODULE_ON_LONG_CMD=4:EchoLink
#QSO_RECORDER=8:QsoRecorder
#ONLINE_CMD=998877
MUTE_RX_ON_TX=1
MUTE_TX_ON_RX=1
#STATE_PTY=/var/run/svxlink/state
#DTMF_CTRL_PTY=/dev/shm/simplex_dtmf_ctrl

[RepeaterLogic]
TYPE=Repeater
RX=Sdr1
TX=Local_Tx
MODULES=ModuleHelp,ModuleParrot
CALLSIGN=26AB1234
SHORT_IDENT_INTERVAL=10
LONG_IDENT_INTERVAL=60
#IDENT_ONLY_AFTER_TX=4
#EXEC_CMD_ON_SQL_CLOSE=500
EVENT_HANDLER=/usr/share/svxlink/events.tcl
DEFAULT_LANG=en_US
RGR_SOUND_DELAY=-1
REPORT_CTCSS=186.2
#TX_CTCSS=SQL_OPEN
MACROS=Macros
#SEL5_MACRO_RANGE=03400,03499
FX_GAIN_NORMAL=0
FX_GAIN_LOW=-12
#QSO_RECORDER=8:QsoRecorder
#NO_REPEAT=1
IDLE_TIMEOUT=2
#OPEN_ON_1750=1000
OPEN_ON_CTCSS=186:500
#OPEN_ON_DTMF=5
#OPEN_ON_SQL=50
#OPEN_ON_SEL5=01234
OPEN_SQL_FLANK=OPEN
#OPEN_ON_SQL_AFTER_RPT_CLOSE=10
IDLE_SOUND_INTERVAL=3000
#SQL_FLAP_SUP_MIN_TIME=500
#SQL_FLAP_SUP_MAX_COUNT=6
#ACTIVATE_MODULE_ON_LONG_CMD=4:EchoLink
#IDENT_NAG_TIMEOUT=15
#IDENT_NAG_MIN_TIME=2000
#ONLINE_CMD=998877
#STATE_PTY=/var/run/svxlink/state
#DTMF_CTRL_PTY=/dev/shm/repeater_dtmf_ctrl

[MultiTx]
TYPE=Multi
TRANSMITTERS=Remote_Tx1,Local_Tx

[Voter]
TYPE=Voter
RECEIVERS=Sdr1,Local_Rx
VOTING_DELAY=200
BUFFER_LENGTH=0
#REVOTE_INTERVAL=1000
HYSTERESIS=50
#SQL_CLOSE_REVOTE_DELAY=500
#RX_SWITCH_DELAY=500
#COMMAND_PTY=/dev/shm/voter_ctrl

[Sdr1]
TYPE=Ddr
#FQ=89300000
#FQ=446193750
FQ=164050000
MODULATION=NBFM
SQL_DET=CTCSS
#VOX_FILTER_DEPTH=20
#VOX_THRESH=500
CTCSS_MODE=2
CTCSS_FQ=71
#CTCSS_SNR_OFFSET=0
#CTCSS_OPEN_THRESH=15
#CTCSS_CLOSE_THRESH=9
#CTCSS_BPF_LOW=60
#CTCSS_BPF_HIGH=270
#SIGLEV_DET=DDR
#SIGLEV_SLOPE=1.54
#SIGLEV_OFFSET=98.10
#SIGLEV_BOGUS_THRESH=120
#TONE_SIGLEV_MAP=100,84,60,50,37,32,28,23,19,8
SIGLEV_OPEN_THRESH=30
SIGLEV_CLOSE_THRESH=10
SQL_START_DELAY=0
SQL_DELAY=0
SQL_HANGTIME=200
#SQL_EXTENDED_HANGTIME=1000
#SQL_EXTENDED_HANGTIME_THRESH=15
#SQL_TIMEOUT=600
DEEMPHASIS=0
#SQL_TAIL_ELIM=300
PREAMP=-4
PEAK_METER=0
DTMF_DEC_TYPE=INTERNAL
DTMF_MUTING=0
DTMF_HANGTIME=40
DTMF_SERIAL=/dev/ttyS0
#DTMF_PTY=/tmp/rx1_dtmf
#DTMF_MAX_FWD_TWIST=8
#DTMF_MAX_REV_TWIST=4
#1750_MUTING=1

WBRX=WbRx1

[WbRx1]
TYPE=RtlUsb
#DEV_MATCH=0
#HOST=localhost
#PORT=1234
#CENTER_FQ=44614375
FQ_CORR=1
GAIN=8.7
SAMPLE_RATE=960000
#SAMPLE_RATE=2400000

[Local_Rx]
TYPE=Local
AUDIO_DEV=alsa:plughw:1
AUDIO_CHANNEL=0
SQL_DET=VOX
SQL_START_DELAY=0
SQL_DELAY=0
SQL_HANGTIME=400
SQL_EXTENDED_HANGTIME=1000
SQL_EXTENDED_HANGTIME_THRESH=15
#SQL_TIMEOUT=600
VOX_FILTER_DEPTH=20
VOX_THRESH=500
#CTCSS_MODE=2
#CTCSS_FQ=71.9
#CTCSS_SNR_OFFSET=0
#CTCSS_OPEN_THRESH=-5
#CTCSS_CLOSE_THRESH=9
#CTCSS_BPF_LOW=80
#CTCSS_BPF_HIGH=180
#SERIAL_PORT=/dev/ttyS0
#SERIAL_PIN=CTS
#SERIAL_SET_PINS=DTR!RTS
#EVDEV_DEVNAME=/dev/input/by-id/usb-SYNIC_SYNIC_Wireless_Audio-event-if03
#EVDEV_OPEN=1,163,1
#EVDEV_CLOSE=1,163,0
#GPIO_PATH=/sys/class/gpio
#GPIO_SQL_PIN=gpio30
#PTY_PATH=/tmp/rx1_sql
#HID_DEVICE=/dev/hidraw3
#HID_SQL_PIN=VOL_UP
#SIGLEV_DET=TONE
SIGLEV_SLOPE=1
SIGLEV_OFFSET=0
#SIGLEV_BOGUS_THRESH=120
#TONE_SIGLEV_MAP=100,84,60,50,37,32,28,23,19,8
SIGLEV_OPEN_THRESH=30
SIGLEV_CLOSE_THRESH=10
DEEMPHASIS=1
#SQL_TAIL_ELIM=300
#PREAMP=6
PEAK_METER=0
DTMF_DEC_TYPE=INTERNAL
DTMF_MUTING=1
DTMF_HANGTIME=40
DTMF_SERIAL=/dev/ttyS0
#DTMF_PTY=/tmp/rx1_dtmf
#DTMF_MAX_FWD_TWIST=8
#DTMF_MAX_REV_TWIST=4
#1750_MUTING=1

[Local_Tx]
TYPE=Local
AUDIO_DEV=alsa:plughw:1
AUDIO_CHANNEL=0
PTT_TYPE=NONE
PTT_PORT=/dev/ttyAMA1
PTT_PIN=RTS
#HID_DEVICE=/dev/hidraw3
#HID_PTT_PIN=GPIO3
#SERIAL_SET_PINS=DTR!RTS
#GPIO_PATH=/sys/class/gpio
#PTT_HANGTIME=1000
TIMEOUT=300
TX_DELAY=0
#CTCSS_FQ=136.5
#CTCSS_LEVEL=9
PREEMPHASIS=0
DTMF_TONE_LENGTH=100
DTMF_TONE_SPACING=50
DTMF_DIGIT_PWR=-15
#MASTER_GAIN=0.0
`

dj1an, 2020-04-27:
>  FQ_CORR=1

Are you sure, this is correct?
Depending on the used SDR Stick, this could be much higher.
The cheap ones need about 40-70 ppm correction.
It may be the case, that you are receiving off Frequency.
Try to evaluate this parameter with a known signal and any SDR Program like GQRX/SDR#/...

73 de DJ1AN, Jan

KayosPi, 2020-04-27:
100% correct, good quality SDR's checked against other equipment. No issues with GQRX, SDR# tested on the laptop as well

I haven't solved it but i have a workaround, rtl_tcp instead of directly reading works fine, i'd rather not use that as the plan eventually is to completely strip out the O/S to a bare minimum. The sound quality and frequency when it does work is superb, even on 6.25khz bandwidth