sm0svx/svxlink #498: SvxReflector sounds

UR3QJW, 2020-05-23:
SvxLink - does not play sounds responsible for the SvxReflector.
If the localization is "en_US" - everything is fine.
If set another localization, for example: "uk_UA" - sounds will not play.

f5vmr, 2020-05-23:
Hi there, this is true, because you will require a structure of language examples that reside at /usr/share/svxlink/sounds . There you will find the en_US folders and the various folders within. The en_US files are rudimentary. But some may have the .txt indicating what they need to be. Some time ago I discovered DSpeech by Dimio, that may still be available, that will require a language either Russian or Ukrainian to build new .wav files. However these .wav files will need trimming with the app Audacity, and exported as Microsoft .wav files. The 12th bit in the .wav files created by DSpeech is incorrect to be read directly in svxlink, but it's corrected by Audacity when "exported" rather than saved. You can remove a long of silences in the sample too, before saving them. It is a long process, I made over 400 samples for the French language, so I know it is possible. Once created you can bundle all the folders into ua_UK, then tar.gz it to a web server. Then sudo wget "webaddress" ua_UK.tar.gz into the sounds folder, then tar -xzvf the file for use as you correctly indicated in your svxlink.conf.
Regards Chris F5VMR G4NAB
 
----------------Le 23/05/2020, à 10:52, UR3QJW a écrit :
SvxLink - does not play sounds responsible for the SvxReflector.
If the localization is "en_US" - everything is fine.
If set another localization, for example: "uk_UA" - sounds will not play.

UR3QJW, 2020-05-23:
I have the necessary files.
But, SvxLink does not find the files here:
/usr/share/svxlink/sounds/uk_UA/
This applies only to the files necessary for SvxReflector.

circool, 2020-05-24:
as i undestand, SvxReflector module does not contain any sound files, as you can see by looking at the original sounds directory.(https://github.com/sm0svx/svxlink/tree/master/src/svxlink/scripts/sounds/mary)
R2ADU.

UR3QJW, 2020-05-24:
The files for the Reflector logic are in the "Core".

sm0svx, 2020-07-27:
SvxLink is able to take a DEFAULT_LANG configuration in both the normal logic section and in the ReflectorLogic section. If a language pack is missing the reflector specific sound clips it will of course not work for that language. A configuration like this should work in that case:

```
[SimplexLogic]
...
DEFAULT_LANG=uk_UA
...

[ReflectorLogic]
...
DEFAULT_LANG=en_US
...
```
Make sure that you are using the [official english langpack](https://github.com/sm0svx/svxlink-sounds-en_US-heather/releases/tag/19.09.99) and that you are running the latest Git master. If running an older version of SvxLink there may be a problem.

Closing this for now. Please reopen if you still think it's a bug in SvxLink.

closed 2020-07-27 by sm0svx