sm0svx/svxlink #503: Feature Request - Configure pull resistors on gpio

Dloranger, 2020-06-19:
It would be helpful to expand the gpio configuration functionality to add the ability to utilize the internal gpio pull up/down resistors on devices when they are created.

oe5hpm, 2020-06-19:
is this really useful ?
what happens at the gpio before creating it ?
for my point of view the "idle-level" has to be defined with a real pull up/down at hardware level.

f5vmr, 2020-06-19:
I've never had a problem.
 
----------------Le 19/06/2020, à 19:02, Daniel Loranger a écrit :
It would be helpful to expand the gpio configuration functionality to add the ability to utilize the internal gpio pull up/down resistors on devices when they are created.

Dloranger, 2020-06-19:
As virtually any true gpio pin is an input and is unknown value at bootup <neither pulled up or down, a very typical situation>, well svxlink isn't running yet anyways so there is no reason to be concerned about its value until svxlink is running.  Only once svxlink is running do we care what is happening at the GPIO and being the GPIO are configured as a prerequisite, they will be in the correct state when we actually care about them.

Dloranger, 2020-06-19:
what caused me to encounter this situation is that I have a signal level adapter (12v->3V) that doesn't have a pull-up on the gpio side.  When the circuit is in active low mode, the blocking diode prevents the radio from pulling the circuit back up so the COS triggers the gpio to go low, but when the COS turns off, the gpio is still stuck in a low state and the PTT stays on.  I envision this will be the same in the active high mode, but I haven't tested it yet.  

Note the circuit always ends up as an active low at the GPIO side.

oe5hpm, 2020-06-20:
could you please post a small drawing of your circuit for better understanding? for example i've no idea what 'COS' is.
A purpose where built-in pull-up would make sense could be possible if you have outside some voltage translator using a fet or transistor.

f5vmr, 2020-06-20:
Hi again

COS is an output from the radio for Squelch such as found in many Motorola GM series. It is normally active High, but make equally be active low in some others, hence the possibility to switch the signals on the diagram I posted.
It is also found on most professional repeater control circuitry but may have other names. So technically from the Receivers point of view there will be COS, RxAudio, and Ground. At the transmitter, Simply PTT, TxAudio and ground.
I have been using Raspberry Pi with SVXLink on repeaters for several years, and never encountered any issues with pull-up resistors in these circuits, and I have 5 Active repeaters.

If any one can solve the problem of stray CTCSS activations when no-one is transmitting to the repeater, Now that would be an interesting development.

73 Chris

oe5hpm, 2020-06-20:
OK, for my understanding the COS can be:
* source output - sourcing 12V if squelch opens - for this a level translator with some transistor is needed with a pull-up to VCCio at the collector/drain
* sink output - the COS is an open-drain

For this use-case a pull-up resistor definitively makes sense.

For some pull-down i don't see actually a use case.

But overall i actually don't see the possibility to turn these pu/d resistors on/off with the kernel sysfs:

https://www.kernel.org/doc/Documentation/gpio/gpio.txt
https://www.kernel.org/doc/Documentation/gpio/sysfs.txt

So most likely we don't have a chance to implement this.
Accessing directly to to SOC registers should be avoided, since we don't want to implement gpio drivers into svxlink i think.

Finally the hard-wired, real life resistor, wins.

Dloranger, 2020-06-21:
I think I may have a method, but its a bit more complicated

_wiringpi_ allows setting of these values as desired, I think there may be some side value in going down this rabbit hole at least as a discussion exercise.  My thought is to slightly retool the setup/teardown scripts to allow use of wiringpi as an alternative to just the use basic bash commands (which are of course most flexible).  I get this won't work for non-rpi install setups, so I am thinking that there could be a try/catch approach that attempts to use wiringpi if available, and if not present, reverts back to the current schema. 

I welcome comments either for or against this approach, but I think this might be something I could come up with in code, I can handle some scripting, but try to stay away from compiled code since it takes me way too long to compile on a pi for each iteration.

oe5hpm, 2020-06-21:
yeah, wiringpi modifies directly the SOC registers.
I think in svxlink we shall only use generic access to gpio or other hardware.

Why not realize your all that with a bash script for starting/stopping the repeater ?
For this nothing within code must be changed.

sm0svx, 2021-08-13:
This is now possible in latest master when using gpiod. However, you need to run libgpiod >= 1.5. Set SQL_GPIOD_BIAS to one of PULLUP, PULLDOWN or DISABLE.

closed 2021-08-13 by sm0svx