sm0svx/svxlink #504: BUG: Svxlink ignore GPIO after TX Timeout (random)

bilson, 2020-07-14:
Hi,

I'm using svxlink with GPIO for Tx and Rx. Svxlink version is master from github.

This is a correct behaviour when TIMEOUT is set to 10 seconds:

> Mon Jul 13 21:44:19 2020: Tx1: Turning the transmitter ON
> Mon Jul 13 21:44:29 2020: *** ERROR: Transmitter Tx1 have been active for too long. Turning it off...
> Mon Jul 13 21:44:37 2020: Tx1: Turning the transmitter OFF

This is what I do in the app:

> 21:44:19: "Transmit" button is pressed.
> 21:44:29: TIMEOUT 10 seconds is reached.
> 21:44:37: The mic is pressed in the app to stop tx and the log show "Tx1: Turning the transmitter OFF". 

This is ok. Tx and Rx works fine after this.

But sometimes, the message "Tx1: Turning the transmitter OFF" don't appears, and Tx1 don't works anymore. Also GPIO COR don't works anymore.

I test the COR GPIO pin status with the command `watch -n0.1 cat /sys/class/gpio/gpio23/value` and transmit with a VHF radio. The pin value change, so, GPIO is working, but svxlink don't read the new value.

The log with bad behaviour is like this:

> Mon Jul 13 21:45:23 2020: Tx1: Turning the transmitter ON
> Mon Jul 13 21:45:33 2020: *** ERROR: Transmitter Tx1 have been active for too long. Turning it off...

Svxlink don't detect when I press the mic in the app to end Tx. This is random. I see acivity from echolink network in log after this, but I can't activate Tx or Rx. 

Restart svxlink is the only solution when this happens.

sm0svx, 2020-07-27:
Are you sure this is connected to the TX timeout?  Does it never happen if the TX timeout does not occur?

Does anything change if you set MUTE_RX_ON_TX=0 in your logic configuration (assuming Simplex). What this test would provide is to see if SvxLink has completely locked up or if you can get the squelch going by setting that configuration variable.

If you trigger an ident command from the console by typing *#, does anything happen while it's locked?

bilson, 2020-07-27:
> Are you sure this is connected to the TX timeout? Does it never happen if the TX timeout does not occur?

Yes, is connected to TX timeout. Don't happen if TX timeout does not occur.

> Does anything change if you set MUTE_RX_ON_TX=0 in your logic configuration (assuming Simplex). What this test would provide is to see if SvxLink has completely locked up or if you can get the squelch going by setting that configuration variable.

I don't try this. I'll make test with this change.
 
> If you trigger an ident command from the console by typing *#, does anything happen while it's locked?

I don't try this. But if I connect to the app I can hear audio from the network. Only GPIO don't respond.

When I'll made the tests, I'll update this issue. Thanks for your reply.

sm3sgp, 2025-04-02:
@bilson Any update on this old issue, or can it be closed?