sm0svx/svxlink #507: Transmitter hangs on open and not reporting squelch closed after upgrage

booyaamps, 2020-07-27:
Hello all! Love the software.
I just built the new source. Repeater has been up for months, configs are good. After the upgrade, any time any site (there are 3 linked via remotetrx) keys up, the squelch never "closes", even though watching the gpio pin indicates that the COR has dropped on the radio side.  Only started happening after the new build. Here's the server config:
###############################################################################
#                                                                             #
#                Configuration file for the SvxLink server                    #
#                                                                             #
###############################################################################

[GLOBAL]
#MODULE_PATH=/usr/lib/arm-linux-gnueabihf/svxlink
LOGICS=RepeaterLogic
CFG_DIR=svxlink.d
TIMESTAMP_FORMAT="%c"
CARD_SAMPLE_RATE=16000
#CARD_CHANNELS=1
#LOCATION_INFO=LocationInfo
#LINKS=LinkToR4

[WeatherLogic]
TYPE=Repeater
RX=WeatherStream
TX=MultiTx
SHORT_IDENT_INTERVAL=0
LONG_IDENT_INTERVAL=0
CALLSIGN=KB2UNV
EVENT_HANDLER=/usr/share/svxlink/events.tcl
DEFAULT_LANG=en_US
OPEN_ON_SQL=100
#OPEN_ON_CTCSS=1050:1000
OPEN_SQL_FLANK=OPEN

[RepeaterLogic]
TYPE=Repeater
RX=Voter
TX=MultiTx
MODULES=ModuleHelp,ModuleMetarInfo,ModuleEchoLink
CALLSIGN=KB2UNV
SHORT_IDENT_INTERVAL=10
LONG_IDENT_INTERVAL=60
IDENT_ONLY_AFTER_TX=0
EVENT_HANDLER=/usr/share/svxlink/events.tcl
DEFAULT_LANG=en_US
RGR_SOUND_DELAY=400
REPORT_CTCSS=91.5
IDLE_TIMEOUT=1
IDLE_SOUND_INTERVAL=0
FX_GAIN_NORMAL=-23
FX_GAIN_LOW=-30
OPEN_ON_SQL=100
#OPEN_ON_CTCSS=91.5:300
OPEN_SQL_FLANK=OPEN
QSO_RECORDER=8:QsoRecorder

[QsoRecorder]
REC_DIR=/usr/share/svxlink/qso_recorder
MIN_TIME=1000
MAX_TIME=3600
SOFT_TIME=300
MAX_DIRSIZE=1024
DEFAULT_ACTIVE=1
TIMEOUT=300
QSO_TIMEOUT=300
#ENCODER_CMD=/usr/bin/oggenc -Q \&quot;%f\&quot; &amp;&amp; rm \&quot;%f\&quot;
ENCODER_CMD=/usr/bin/oggenc \"%f\" && rm \"%f\" 

[Voter]
TYPE=Voter
RECEIVERS=MainSite,SecondSite,KD2ITK
VOTING_DELAY=10
BUFFER_LENGTH=0
REVOTE_INTERVAL=100
HYSTERESIS=50
#SQL_CLOSE_REVOTE_DELAY=200
RX_SWITCH_DELAY=10
#COMMAND_PTY=/dev/shm/voter_ctrl

[MultiTx]
TYPE=Multi
TRANSMITTERS=MainSite,SecondSite,KD2ITK

sm3sgp, 2020-07-27:
What software revision are you using?
What squelch types?
It should be easy to identify what specific receiver squelch that stays open.

booyaamps, 2020-07-27:
Hi!
  I cloned the current git repository and build via the instructions.
Before building, i did remove/purge/backup the existing version (17).
All stations are using GPIO squelch. Also, any station is producing the
same issue. I think i've narrowed it down to an issue with svxlink-gpio but
it seems to be reading the gpio events properly.

sm0svx, 2020-11-27:
So did you find the problem?

What may not be obvious is that SvxLink need a constant flow of audio samples since some timing, like squelch close hangtime etc, depend on that. So if you have an odd setup where audio samples is not coming in constantly, that may be your problem.

booyaamps, 2020-11-27:
Thanks for the reply!
Yes, this was an odd upgrade relic. I guess before upgrading i never
stopped the svxlink_gpio service (we were using gpio for tx at the time,
remotetrx site, and have since moved to serial pins) and that service never
got upgraded with svxlink itself. Purging and rebuilding fixed the issue.

sm0svx, 2020-11-27:
ok that's good! Please close issues that is resolved to help us keep the issue tracker clean.

closed 2020-11-27 by sm0svx