sm0svx/svxlink #509: Directory server connection over proxy server is not closed correctly

dh1dm, 2020-08-14:
The connection to the directory server over a proxy server can be closed in two ways. Either the directory server has no more data to send and closes the TCP connection. The proxy server then sends a PROXY_MSG_TCP_CLOSE message which correctly puts the client into TCP_STATE_DISCONNECTED state. The second way is the tcpClose function of the Proxy class. This sends a PROXY_MSG_TCP_CLOSE to the proxy server and puts the client into TCP_STATE_DISCONNECTING state (disconnect pending). In this situation, at least the proxy server written by Bob (http://pe1chl.nl/Softw/elproxy.tar.gz) does not send any further response and there is no way to finish the pending disconnect on the client side resulting in a state transition to TCP_STATE_DISCONNECTED state. No further directory server connections can be established in this situation.
Two possible solutions exist to overcome this problem. Either the tcpClose function immediately enters the TCP_STATE_DISCONNECTED state or the proxy server acknowledges the PROXY_MSG_TCP_CLOSE request with a PROXY_MSG_TCP_CLOSE reply.

pe1chl, 2020-08-14:
Note that my proxy server has exactly the same behavior as the "official" Echolink proxy server written in Java.
When there is a problem with this behavior, I suggest that it is fixed in svxlink rather than in the proxy...

For completeness: there also is a 60 second timeout on directory server sessions, when it is exceeded the session is just closed by the proxy and it sends the PROXY_MSG_TCP_CLOSE regardless of the incomplete data.
This is also the same as in the "official" proxy.

dh1dm, 2023-07-28:
The proxy always sends a PROXY_MSG_TCP_CLOSE reply on client disconnect. This requires a multi-state handling,. However, the client may remain in disconnect pending state in case of an unreliable/broken TCP connection. This issue should rather be addressed by improved connection timers. It can cause a permanently blocking EchoLink module, but is hardly reproducible.