sm0svx/svxlink #513: SVX not playing any wav while command recieved by DTMF.

ok2zo, 2020-09-30:
Hello, having this issue longer time.
When send command directly from console by keyboard, all played fine, PTT works. When send via radio by dtmf, can see decoded fine, shows ptt on, playing, ptt off but real output none..
I thought it was some corruption on rpi instalation. Now installed on another machine and the issue is the same..
Echolink audio stream, IDs, all works fine.. Commands, like * or  help etc not. Can't find any logic in that. Could you help, please?

sm0svx, 2020-11-27:
Still a problem?  Have your tried asking at the [groups.io discussion forum](https://groups.io/g/svxlink)?

ok2zo, 2020-11-28:
Hello Tobias.

Finnaly I have found the fault. I had to uncomment the parameter EXEC_CMD_ON_SQL_CLOSE=0.
Not sure, why is it necessary for DR-2X repeater, but now it works well.  

I have noticed, when testing, that somebody keyed repeater after command and I've heard some words passing through. So tried above and it worked..

Best regards,

Libor

  ----- Original Message ----- 
  From: Tobias Blomberg 
  To: sm0svx/svxlink 
  Cc: ok2zo ; Author 
  Sent: Friday, November 27, 2020 6:21 PM
  Subject: Re: [sm0svx/svxlink] SVX not playing any wav while command recieved by DTMF. (#513)

  Still a problem? Have your tried asking at the groups.io discussion forum?

closed 2020-11-28 by sm0svx