sm0svx/svxlink #514: Docker image for svxlink

joergklein, 2020-10-06:
I wrote a docker image for svxlink. The operating system is CentOS 8. Only the qtel client is working. I get an error message when I start svxlink. How th configure a the repeater and the rest of svxlink? Qtel works without any errors.

The GitHub Repo:
https://github.com/joergklein/docker-svxlink

The Docker image
https://hub.docker.com/r/joergklein/svxlink

The Error Message:
```
Starting logic: SimplexLogic
Loading RX: Rx1
*** ERROR: Open capture audio device failed: No such file or directory
*** ERROR: Could not open audio device for receiver "Rx1"
*** ERROR: Could not initialize RX "Rx1"
*** ERROR: Could not initialize Logic object "SimplexLogic". Skipping...
*** ERROR: No logics available. Bailing out...
```

dl1hrc, 2020-10-06:
The error message shows, that you do not have any usable sound card installed on your system or the sound card does not have a capture device with id=0 (which is the default).

check AUDIO_DEV=alsa:plughw:0 in [Rx1]-section

joergklein, 2020-10-06:
I don't know how to check AUDIO_DEV=alsa:plughw:0 in [Rx1]-section

```
arecord -l
**** List of CAPTURE Hardware Devices ****
card 1: PCH [HDA Intel PCH], device 0: CX20751/2 Analog [CX20751/2 Analog]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

--
```
sh-4.4# arecord -L
surround21
    2.1 Surround output to Front and Subwoofer speakers
surround40
    4.0 Surround output to Front and Rear speakers
surround41
    4.1 Surround output to Front, Rear and Subwoofer speakers
surround50
    5.0 Surround output to Front, Center and Rear speakers
surround51
    5.1 Surround output to Front, Center, Rear and Subwoofer speakers
surround71
    7.1 Surround output to Front, Center, Side, Rear and Woofer speakers
null
    Discard all samples (playback) or generate zero samples (capture)
sysdefault:CARD=PCH
    HDA Intel PCH, CX20751/2 Analog
    Default Audio Device
front:CARD=PCH,DEV=0
    HDA Intel PCH, CX20751/2 Analog
    Front speakers

```
-
```
sh-4.4# cat /proc/asound/cards
 0 [HDMI           ]: HDA-Intel - HDA Intel HDMI
                      HDA Intel HDMI at 0xe1330000 irq 52
 1 [PCH            ]: HDA-Intel - HDA Intel PCH
                      HDA Intel PCH at 0xe1334000 irq 53
```

gustavc39, 2020-10-06:
Hi Joerg,

I don't know how to check AUDIO_DEV=alsa:plughw:0 in [Rx1]-section
>

That parameter is located in the main configuration file, which by default
is /etc/svxlink/svxlink.conf

You may also want to check user permissions to verify that the user running
svxlink has access to the sound system.

Regards.

Gustavo - LU7WA

joergklein, 2020-10-06:
This is my entry in the Dockerfile

`RUN useradd -r -g daemon -G svxlink -c "svxlink" svxlink`

Please check my Dockerfile

dl1hrc, 2020-10-06:
can it be that your first sound card ([0] - HDA-Intel - HDA Intel HDMI) doesn't have a capture device? You can check it e.g. with the alsamixer

joergklein, 2020-10-07:
I found the solution.
change in /etx/svxlink/svxlink.conf `AUDIO_DEV=alsa:plug hw:0` to `AUDIO_DEV=alsa:plug hw:1`

`

closed 2020-10-07 by joergklein