sm0svx/svxlink #522: problem with alsa lib pcm_hw.c:1401

mnjagadeesh, 2020-11-18:
I am using Linaro 13.09 os on radxa rock. I did source svxlink installation. Everything is working fine. But alsa is seem to be and issue. am getting "ALSA lib pcm_hw.c:1401:(_snd_pcm_hw_open) Invalid value for card"

starting logic: SimplexLogic
Loading RX: NONE
Loading TX: Tx1
ALSA lib pcm_hw.c:1401:(_snd_pcm_hw_open) Invalid value for card
Loading module "ModuleHelp" into logic "SimplexLogic"
	Found /usr/local/lib/arm-linux-gnueabihf/svxlink/ModuleHelp.so
	Module Help v1.0.0 starting...

if I use AUDIO_DEV=oss:/dev/ttyUSB0 I do not see this error but no sound coming through.

With `speaker-test -c2 -D plughw:3,0` works well. Any pointer to check?

mnjagadeesh, 2020-11-26:
I had to run svxlink with elevated permission.

closed 2020-11-26 by mnjagadeesh

sm0svx, 2020-11-27:
Firstly, don't use this issue tracker to ask questions. It's only meant to report bugs and feature requests. Instead use the [discussion group at groups.io](https://groups.io/g/svxlink) if your're not sure it's a bug. Anyway, thanks for closing the issue when you solved the problem! Not everyone bothers to do that.

For security reasons, don't run SvxLink with elevated permissions. Instead, identify which groups you need to add the SvxLink user to in order for it to gain access to the hardware devices it needs. For Alsa audio it need access to the device nodes under /dev/snd so use a command like ```$ ls -l /dev/snd/``` and you will probably see that the SvxLink used need to be a member of the "audio" group or something similar.

mnjagadeesh, 2020-11-28:
Got it. Thanks for replying.