sm0svx/svxlink #523: Feature Request - Direct access to ONLINE_CMD function

jcacioppo59, 2020-11-27:
I am using svxlink with the Open Repeater Project and need your help with the ONLINE_CMD
My computer skills are novice at best
Is it possible to link control of the ONLINE_CMD to a GPIO state instead of a DTMF function?
I have a specific GPIO in mind such that it could be included into the code directly.
I appreciate any help or direction you can offer.

sm0svx, 2020-11-27:
Implementing direct support for setting online/offline using GPIO is a little bit too specific I think. I have instead made it possible to use a PTY to send commands to SvxLink. This now make it possible to set online/offline from a script that monitors the GPIO pin.

In your logic core configuration section (e.g. RepeaterLogic) add:
```
COMMAND_PTY=/dev/shm/repeater_logic_ctrl
```
Then you can set your repeater offline by executing this in your shell:
```
echo 'CFG RepeaterLogic ONLINE 0' > /dev/shm/repeater_logic_ctrl
```
To set it online again execute:
```
echo 'CFG RepeaterLogic ONLINE 1' > /dev/shm/repeater_logic_ctrl
```
It's up to you to write the script that monitors the GPIO pin and send the commands above. If you need any help with that try asking at the [groups.io discussion forum](https://groups.io/g/svxlink).

For more information on the new functionality have a look at commit b27724df22e3e3cd713f08c28acb9e4fbe8d86fc.

closed 2020-11-27 by sm0svx

alleghenycreative, 2021-04-01:
Thanks for your work on this Tobias. While we wait for this to be implemented in a future release (since we only build ORP builds off release versions), I have started to add in some test code to utilize other PTYs in ORP like the dtmf_pty. I've successfully created a test form that can send dtmf commands to a selected logic via the web ui using ajax commands. It is in my dev code base 2.2.x and early work on 3.0.x. The PTYs should prove to be very useful down the road.