sm0svx/svxlink #526: SvxLink does not compile against libsigc++ > 2.7.1

najdanovicivan, 2020-12-17:
Starting from version 2.7.2 I cannot compile the SvxLink

I'm getting the following errors:
[sigc++ .txt](https://github.com/sm0svx/svxlink/files/5706417/sigc%2B%2B.txt)

I see that latest version 2 of libsigc++ is 2.99.9

There is also a version 3.0.6 but I think this will need a lot of changes to be done in order to use it

sm0svx, 2020-12-19:
I'm not sure that this is caused by sigc (only). It build find on Arch (libsigc++ 2.10.6) and Fedora 33 (libsigc++20-devel-2.10.4). Or maybe it's a problem with that exact version of sigc that you are using.

It could possibly also be caused by a compiler problem if your compiler was recently updated. Or maybe because you are using musl libc?

najdanovicivan, 2020-12-19:
Im compiling it using OpenWrt SDK . Os yeah it’s musl based. I’ve sucessfully built it with libsigc++ 2.7.1 and it being used for YU1INO-R at the moment runing from the X86_64 bit system

I want to submit PR now to OpenWrt packages repo but OpenWrt is missing gsm and libsigc++ so I need to add those as well but I want to add the most recent versions which will work

Here is my OpenWrt feed https://github.com/najdanovicivan/openwrt-packages  

I also checked the libsigc++ repo i saw that version 2.7.2 is marked as unsable. I’ll try with 2.10.6 to see how it goes.  

I wonder how much work will it take to upgrade libsigc++ to version 3

I also intend to add full luci support for OpenWrt so that SvxLink is completely configurable wis web ui

Having SvxLink on OpenWrt makes it possible to run it on any supported router with USB soundcard

sm0svx, 2020-12-20:
There is no need to upgrade to sigc++ 3 until it is widely available on all common distributions and as long as sigc++ 2 is available it will not be a problem. But sooner or later I will need to make the move. Maybe it's not a big problem unless there are major API changes.

najdanovicivan, 2020-12-29:
I've successfully compile svxlink with libsigc++ 2.10.6. There was an issue with automake so I had to compile libsigc++ with meson.

closed 2020-12-29 by najdanovicivan