sm0svx/svxlink #527: Feature Requset - Local Transmitter - PTT_TYPE - MuteInput

najdanovicivan, 2020-12-29:
I've got the cheap chinese usb audio adapter like this one

https://www.dhgate.com/product/wholesale-cubieboard2-usb-7-1ch-sound-card/382555428.html

It's all fine to use hidraw for squelch input.

The issue with these is that CM119 is epoxy glued on board so there is no way to access the GPIO pin to use it for PTT

What's interesting about this card is when input is muted the red led will be lit.

This makes it possible to use the led for triggering PTT in Simplex mode.

As it's being used in simplex mode it won't be possible to use the input at the same time as output so what I need is an option to mute the input when PTT is active. and unmute it back once the transmission stops

najdanovicivan, 2021-01-09:
Any info on this one. I just basically need to be able to mute Alsa capture input on TX ?

sm0svx, 2021-08-13:
SvxLink does not manipulate audio device settings at all like volume and muting so you need to do this outside of SvxLink. One way would be to call the external application "amixer" from the transmit function in Logic.tcl. Another way is to use PTT_TYPE=PTY and write an external script that listen on the PTY to mute/unmute the audio device on tx/rx.

closed 2021-08-13 by sm0svx