sm0svx/svxlink #530: New Bug received Today on 10.01.2021 Svxlink Version newest

DH3PAE, 2021-01-10:
Should now be fixed in latest master

_Originally posted by @sm0svx in https://github.com/sm0svx/svxlink/issues/529#issuecomment-757373150_

SimplexLogic: digit=#
Rx1: The squelch is CLOSED (712)
DTMF command received in module EchoLink: 
DL8BBY: EchoLink QSO state changed to DISCONNECTED
Rx1: The squelch is OPEN (2219)
Rx1: The squelch is CLOSED (75)
Tx1: Turning the transmitter ON
Tx1: Turning the transmitter OFF
SimplexLogic: Module timeout: EchoLink
SimplexLogic: Deactivating module EchoLink...
Tx1: Turning the transmitter ON
Tx1: Turning the transmitter OFF
Rx1: The squelch is OPEN (1336)
SimplexLogic: digit=3
SimplexLogic: digit=1
SimplexLogic: digit=#
Rx1: The squelch is CLOSED (939)
dtmfCmdReceivedWhenIdle
DTMF command received in module MetarInfo: 1
https://aviationweather.gov/adds/dataserver_current/httpparam?dataSource=metars&requestType=retrieve&format=xml&hoursBeforeNow=3&mostRecent=true&stationString=EDDK
Rx1: The squelch is OPEN (1796)
requesting XML metar version from https://aviationweather.gov/
XML-METAR: EDDK 101320Z 15004KT 9000 SCT009 BKN042 03/01 Q1024 NOSIG
UTC: 13:31:54 daytime saving:0 vs 13:20:0
metar "EDDK 101320Z 15004KT 9000 SCT009 BKN042 03/01 Q1024 NOSIG"
announce_airport EDDK
metreport_time 1320
wind 150 04 unit_kts
visibility 9 unit_kms
clouds sct 900
clouds bkn 4200
temperature 3
dewpoint 1
qnh 1024
trend nosig_long 
Rx1: The squelch is CLOSED (960)
Tx1: Turning the transmitter ON
svxlink: /home/dh3pae/svxlink/src/async/audio/AsyncAudioProcessor.cpp:322: void Async::AudioProcessor::writeFromBuf(): Assertion `(written >= 0) && (written <= buf_cnt)' failed.
Abgebrochen (Speicherabzug geschrieben

sm0svx, 2021-01-10:
Please try the latest git master which I just pushed. I don't know if it will solve your problem but it's worth a try.

sm3sgp, 2025-03-16:
@DH3PAE Can this issue be closed?

DH3PAE, 2025-03-17:
yes

Diese E-Mail wurde von Avast-Antivirussoftware auf Viren geprüft.
www.avast.com

closed 2025-03-17 by sm3sgp