sm0svx/svxlink #531: BUG: ModuleMetarInfo. Wind gusts

ur3qjw, 2021-01-26:
Hi,
MetarInfo module, incorrectly plays the values of wind gusts.

metar "... TEMPO 23012G20MPS"
> "wind 230 12 unit_mps 0m unit_mps"
metar "... TEMPO 26012G18MPS"
> wind 260 12 unit_mps 8m unit_mps

XML-METAR: UKHH 261130Z 20005MPS CAVOK 04/00 Q1009 R25/290076 TEMPO 23012G20MPS
UTC: 11:55:59 daytime saving:0 vs 11:30:0
metar "UKHH 261130Z 20005MPS CAVOK 04/00 Q1009 R25/290076 TEMPO 23012G20MPS"
announce_airport UKHH
metreport_time 1130
wind 200 05 unit_mps
say cavok_long 
temperature 4
dewpoint 0
qnh 1009
runwaystate runway 25 wet_or_water_patches contamination 51 to 100 percent deposit_depth less_than 1 unit_mm  friction_coefficient 0.76
say tempo
wind 230 12 unit_mps 0m unit_mps

XML-METAR: UKBB 261130Z 23011MPS 9999 BKN024 03/M03 Q1001 R18L/190055 TEMPO 26012G18MPS
UTC: 11:53:52 daytime saving:0 vs 11:30:0
metar "UKBB 261130Z 23011MPS 9999 BKN024 03/M03 Q1001 R18L/190055 TEMPO 26012G18MPS"
announce_airport UKBB
metreport_time 1130
wind 230 11 unit_mps
visibility more_than 10 unit_kms
clouds bkn 2400
temperature 3
dewpoint -3
qnh 1001
runwaystate runway 18 left damp contamination 51 to 100 percent deposit_depth less_than 1 unit_mm  friction_coefficient 0.55
say tempo
wind 260 12 unit_mps 8m unit_mps

dl1hrc, 2021-01-26:
Hi Volodymyr,

thank you for your bug report, could you check this please:
https://github.com/dl1hrc/svxlink/tree/metarbug-26012021

mni 73s de Adi / DL1HRC

ur3qjw, 2021-01-27:
Thanks Adi!
I use this. But so far no such metar-reports have been received.
I will let you know how the information appears.

Or, I can somehow simulate this report ?
metar "UKHH 261130Z 20005MPS CAVOK 04/00 Q1009 R25/290076 TEMPO 23012G20MPS"

And.. In file "MetarInfo.tcl", how to change the playback sequence ?
From: "UKKK => airport".
On: "airport => UKKK".

dl1hrc, 2021-01-27:
You can simulate it by doeing some changes in your ModuleMetarInfo.conf. You will need a small webserver (localhost may be ok) and the Metarfile UKHH in the /download/ folder. UKHH has exactly 2 lines in this format:

```
2021/01/26 15:20
UKHH 261130Z 20005MPS CAVOK 04/00 Q1009 R25/290076 TEMPO 23012G20MPS
```

The date and time is important and will be checked when you try to call the Metar (max 2 hours difference to current time). Then change the ModuleMEtarInfo.conf:

```

[ModuleMetarInfo]
....
TYPE=TXT
SERVER=your_private_webserver.net
LINK=/download/
...
LONGMESSAGES=1
REMARKS=1
DEBUG=1
AIRPORTS=UKHH,ESSB,EDDP,SKSM,EDDS,EDDM,EDDF,KJAC,KTOL

```

The announcement sequence is defined in MetarInfo.tcl  / procedure announce_airport,
here the original file:

```
proc announce_airport {icao} {
  global langdir;
  if [file exists "$langdir/MetarInfo/$icao.wav"] {
    playMsg $icao;
  } else {
    spellWord $icao;
  }
  playSilence 100;
  playMsg "airport";
}
```

maybe this chnge will solve your request:

```
proc announce_airport {icao} {
  global langdir;
  playMsg "airport";
  playSilence 100;
  if [file exists "$langdir/MetarInfo/$icao.wav"] {
    playMsg $icao;
  } else {
    spellWord $icao;
  }
}
```
Remember to do it according the locale concept, otherwise you will loose your changes on next "make install"

mni 73s de Adi / DL1HRC

ur3qjw, 2021-01-27:
Thanks Adi!
Your changes are working. I received a fresh Metar report.
The simulation also showed that everything is working.

metar "UKDE 271830Z 12007G12MPS 080V140 CAVOK 07/03 Q0998 R20/220567 TEMPO 14007G12MPS"
"wind 140 07 unit_mps 12 unit_mps"

closed 2021-01-27 by sm0svx