sm0svx/svxlink #533: Audio filter is not working

DF2HS, 2021-01-29:
Hello

As far as I know, the audio filter has been added to svxlink, so no manual compilation is required anymore.

See also:
https://forum.funk-telegramm.de/thread/579-audiofilter-f%C3%BCr-svxlink/

But when I add  a filter in my RX section, nothing really happens.

For example:
AUDIO_FILTER=BpBu4/50-2700

Or just the given example from the page above:
AUDIO_FILTER=LpBe1/1000

Nothing happens. The audio is not filtered.

Can please anyone tell me what I am doing wrong?

73
Sandy DF2HS

dl1hrc, 2021-01-29:
Hi,
no, there's no param AUDIO_FILTER in the recent trunk. As far as I remember I've created a branch but it didn't found the way into the trunk. The filter audio parameters are hardcoded in the Rx/Tx-classes.
mni 73s de Adi / DL1HRC

DF2HS, 2021-01-29:
Would you please be so kind and check the given code from the forum page, because it isn't compiling with the current svx version.

So when I understand you right, it is possible to choose filters before compiling which are locked and cannot be adjusted without recompiling. That's a bit sad.

Is it possible to generate a more user friendly method?

Thank you very much.

dl1hrc, 2021-01-29:
You may try this brunch. I've merged with master rigth now.
https://github.com/dl1hrc/svxlink/tree/audiofilter
Then you can define a param AUDIO_FILTER. But it's some time ago I have played with that and I couldn't test it, please give a short feedback.

mni 73s de Adi / DL1HRC

DF2HS, 2021-01-30:
Hi Adi

I followed your link and then cloned this link:

[](https://github.com/dl1hrc/svxlink.git)

In my remotetrx.conf I added

`AUDIO_FILTER=BpCh12/-0.1/300-3500`

This line is getting ignored and the audio is not getting filtered.

dl1hrc, 2021-01-30:
Have you really installed the complete audiofilter-branch?
Just tried here: the param isn't ignored (svxlink.conf or/and remotetrx.conf) and the audio was limited.
73s de Adi / DL1HRC

DF2HS, 2021-01-31:
So when I try this:

`pi@raspberrypi:~ $ git clone https://github.com/dl1hrc/svxlink/tree/audiofilter
Klone nach 'audiofilter' ...
fatal: repository 'https://github.com/dl1hrc/svxlink/tree/audiofilter/' not found
`

You see, that is not working.

I can only download this, but that is not including the updated files I think:

https://github.com/dl1hrc/svxlink.git

73
Sandy DF2HS

DF2HS, 2021-01-31:
I manually edited the three files which you changed and compiled again. Now it works.

The audio gets filtered and there is a info line:

`Using configuration file: /etc/svxlink/remotetrx.conf
--- Using sample rate 48000Hz
Setting up trx "NetUplinkTrx10mver"
RX: Rx1
+++ Init Audiofilter=BpCh12/-0.1/300-2500
TX: Tx1
`

Seems to work. Maybe it will find its way into the master trunk.

73
Sandy DF2HS

dl1hrc, 2021-01-31:
git clone https://github.com/dl1hrc/svxlink.git
cd svxlink
git checkout audiofilter
cmake,make....and so on

DF2HS, 2021-02-01:
Okay, I didn´t know about the checkout command.
I just tried it again and it works.

```
Starting logic: RepeaterLogic
Loading RX: Voter
	Adding receiver: Rx1
+++ Init Audiofilter=BpBu4/300-2700
	Adding receiver: RemoteRX
------ Opus decoder parameters ------
Gain       = 0dB
--------------------------------------
Loading TX: MultiTx
	Adding transmitter: Tx1
+++ Init Audiofilter=BpCh12/-0.1/300-2700
	Adding transmitter: StreamingTx

```

Thank you!

DF2HS, 2021-02-02:
Can you merge that with the master branch?
And, I saw the parameter for the limiter. Would be nice to have them in the config, too.

Thanks alot!

73
Sandy DF2HS

DF2HS, 2023-04-11:
> You may try this brunch. I've merged with master rigth now. https://github.com/dl1hrc/svxlink/tree/audiofilter Then you can define a param AUDIO_FILTER. But it's some time ago I have played with that and I couldn't test it, please give a short feedback.
> 
> mni 73s de Adi / DL1HRC

Hi Adi

I tried your modification with the current svx version and therefore edited the LocalRxBase.cpp and LocalTx.cpp manually and added the following code:

`string filterdesc;
  if (cfg().getValue(name(), "AUDIO_FILTER", filterdesc))
  {
    cout << "+++ Init Audiofilter=" << filterdesc << endl;
    AudioFilter *additional_audiofilter = new AudioFilter(filterdesc);
    prev_src->registerSink(additional_audiofilter, true);
    prev_src = additional_audiofilter;
  }`

... but I get an error like this:

`[ 62%] Building CXX object svxlink/trx/CMakeFiles/trx.dir/LocalTx.cpp.o
/home/pi/svxlink/src/svxlink/trx/LocalTx.cpp: In member function ‘virtual bool LocalTx::initialize()’:
/home/pi/svxlink/src/svxlink/trx/LocalTx.cpp:483:11: error: no match for call to ‘(Async::Config) ()’
  483 |   if (cfg().getValue(name(), "AUDIO_FILTER", filterdesc))
      |           ^
make[2]: *** [svxlink/trx/CMakeFiles/trx.dir/build.make:199: svxlink/trx/CMakeFiles/trx.dir/LocalTx.cpp.o] Fehler 1
make[1]: *** [CMakeFiles/Makefile2:2542: svxlink/trx/CMakeFiles/trx.dir/all] Fehler 2
make: *** [Makefile:171: all] Fehler 2
`

Make only works if I comment the code section out.

I did some more tests:

Audiofilter are working on RX after modification, but not on TX (because I had to comment them out).

Could someone assist?

73
Sandy DF2HS

dl1hrc, 2023-04-11:
please pull the branch "audiofilter" and recompile.
Mni 73s de Adi / DL1HRC

DF2HS, 2023-04-11:
> 

hm okay I copied the files to the origin folder after checkout your files and recompiled, now it works. Thanks. Hopefully it gets its way to the master branch.

73
Sandy DF2HS

dl1hrc, 2023-04-12:
Why? The audiofilter branch is even with the master. There is no need to do this.

DF2HS, 2023-04-12:
But why do I have to modify the files then each time?I always download the files from sm0svx and modify them for the audiofilter because I don't know if your files are always up-to-date, but maybe I just don't understand enough how this all works 😂

dl1hrc, 2023-04-12:
Genau das mußt du ja nicht machen, wenn Du den Branch direkt nutzt:

```
git clone https://github.com/dl1hrc/svxlink.git
cd svxlink
git checkout audiofilter
cd src
mkdir build
cd build
cmake ...
```
bzw. wenn Du das schon mal gemacht hast einfach nur ein "git pull" um die Änderungen nachzuziehen.

> Ist denn "dein" Svx immer aktuell?

Diesen Branch halte ich so gut es geht aktuell zum Master von Tobias, so dass Du alle Funktionen dessen verfügbar sind. Es kann auch mal sein, dass ich nicht gleich dazu komme aber meistens passt es. Den Stand kannst Du checken indem Du  im git nachschaust:

[https://github.com/dl1hrc/svxlink/tree/audiofilter](https://github.com/dl1hrc/svxlink/tree/audiofilter)

Da steht im Moment "This branch is [10 commits ahead of sm0svx:master." Der Branch ist also 10 commits vor dem Master, hat also alles vom Master und zusätzlich die entsprechenden Erweiterungen.
Wenn die Erweiterung irgendwann mal in den Master übernommen werden soll ist es schon erforderlich diesen Branch zu testen/nutzen. Wenn man händisch hin- und herkopiert kann man das irgendwann nicht mehr nachvollziehen.

73s Adi / DL1HRC

DF2HS, 2023-04-12:
> bzw. wenn Du das schon mal gemacht hast einfach nur ein "git pull" um die Änderungen nachzuziehen.

Ja, stimmt.

> > Ist denn "dein" Svx immer aktuell?
Leider nein, daran arbeite ich noch. 

> Diesen Branch halte ich so gut es geht aktuell zum Master von Tobias, so dass Du alle Funktionen dessen verfügbar sind. 
Okay, ja ich war mir halt nicht sicher, bzw. ging davon aus, dass "dein" Svx hier mehr eine Experimentierwiese ist, um neue Funktionen zu testen. Quasi eine Betaversion. Daher nahm ich bislang immer die von Tobias.

>Wenn die Erweiterung irgendwann mal in den Master übernommen werden soll ist es schon erforderlich diesen Branch zu testen/nutzen. 
Da hast du natürlich Recht. Wobei es ja aus meiner laienhaften Sicht auch "nur" diese zwei, drei Dateien sind, aber irgendwie wohl auch nicht. Setze die Tage eh ein neues System auf, dann nehme ich mal "dein" Svx.

73
Sandy DF2HS

DF2HS, 2023-07-12:
Du meintest mal, du bräuchtest eine Rückmeldung, wie deine Software so spielt. Seit ein paar Monaten ist "deine" Svx Version bei DB0WAG installiert und seit ein paar Tagen bei DB0HHH 2m im Testbetrieb. Bis jetzt keine größeren Schwierigkeiten. Weitere Repeater werden folgen.

dl1hrc, 2023-07-12:
Moin Sandy, das freut mich zu hören/lesen. Vielen Dank. Mni 73s de Adi / DL1HRC