sm0svx/svxlink #535: Svxserver connection suddenly break while Echolink incoming connection

dl7ata, 2021-02-13:
I don't know if it is a bug or a feature (missing a parameter):
My audio stream into the svxserver break in the same moment with an incoming echolink connection. After disconnect all others can hear me again. There is nothing in the log:

> 13.02.2021 11:31:07: Logic_DL7ATA: Talker stop on TG #2621: DC7XJ
13.02.2021 11:31:07: Logic_Ostlink: Talker stop on TG #81: DB0TGO
13.02.2021 11:31:07: 70cmTx: Turning the transmitter OFF
13.02.2021 11:31:07: 2mTx: Turning the transmitter OFF
13.02.2021 11:31:08: Spurious audio packet received from 44.137.75.243
13.02.2021 11:31:08: Incoming EchoLink connection from DL7ATA (Frank) at 44.137.75.243
13.02.2021 11:31:10: Spurious audio packet received from 44.137.75.243
13.02.2021 11:31:15: Incoming EchoLink connection from DL7ATA (Frank) at 44.137.75.243
13.02.2021 11:31:15: ### Register incoming station, count 1 of 3 possible number of connects
13.02.2021 11:31:15: SimplexLogic: Activating module EchoLink...
13.02.2021 11:31:15: DL7ATA: Accepting connection. EchoLink ID is 595177...
13.02.2021 11:31:15: DL7ATA: EchoLink QSO state changed to CONNECTED
13.02.2021 11:31:15: Ruft ZVEI 03580
13.02.2021 11:31:15: --- EchoLink info message received from DL7ATA ---
13.02.2021 11:31:15: Station DL7ATA
13.02.2021 11:31:15: Frank
13.02.2021 11:31:15: 
13.02.2021 11:31:15: samsung SM-A705FN User
13.02.2021 11:31:15: 
13.02.2021 11:31:15: DL7ATA is running EchoLink for Android on a samsung SM-A705FN, with Android version 10.
13.02.2021 11:31:15: 70cmTx: Turning the transmitter ON
13.02.2021 11:31:27: 70cmTx: Turning the transmitter OFF
13.02.2021 11:31:28: 70cmRx: The squelch is OPEN
13.02.2021 11:31:31: 70cmRx: The squelch is CLOSED
13.02.2021 11:31:31: 70cmTx: Turning the transmitter ON
13.02.2021 11:31:32: 70cmTx: Turning the transmitter OFF
13.02.2021 11:31:40: 70cmRx: The squelch is OPEN
13.02.2021 11:31:42: 70cmRx: The squelch is CLOSED
13.02.2021 11:31:42: 70cmTx: Turning the transmitter ON
13.02.2021 11:31:42: 70cmTx: Turning the transmitter OFF
13.02.2021 11:31:46: 70cmTx: Turning the transmitter ON
13.02.2021 11:31:49: 70cmTx: Turning the transmitter OFF
13.02.2021 11:31:51: DL7ATA: EchoLink QSO state changed to BYE_RECEIVED
13.02.2021 11:31:51: DL7ATA: EchoLink QSO state changed to DISCONNECTED
13.02.2021 11:31:51: SimplexLogic: Deactivating module EchoLink...
13.02.2021 11:31:51: 70cmTx: Turning the transmitter ON
13.02.2021 11:31:58: 70cmTx: Turning the transmitter OFF
13.02.2021 11:31:58: 70cmRx: The squelch is OPEN
13.02.2021 11:31:58: Logic_Ostlink: Talker start on TG #81: DB0TGO

This phenom appears in two completely different systems, one is pure internet with round about 12 clients, the other is a hamnet interlink with 3 repeaters.

dl1hrc, 2021-02-14:
Hi Frank,
maybe you have enabled MUTE_LOGIC_LINKING=1 in your ModuleEchoLink.conf. 
I've not tried it but I guess it's a new switch to disconnect the svxreflector-network from a node that is accessed by an EchoLink connection.
mni 73s de Adi / DL1HRC

dl7ata, 2021-02-14:
Hi Adi,
there is no _MUTE_LOGIC_LINKING_ in my CONF. The only MUTE I found is `MUTE_FIRST_TX_LOC=0 `  in a reflector logic. But what happens when   _MUTE_LOGIC_LINKING_  is missing? Maybe I give it a try by explicit setting it to 0.

dl7ata, 2021-02-14:
Another test:
In the moment when I activate manually the EchoLinkModule, without any further dialing, my client stop sending audio to the svxserver, no other client can hear me. I put all these params into my svxlink.conf:
`MUTE_FIRST_TX_LOC=0
MUTE_LOGIC_LINKING=0  
MUTE_FIRST_TX_REM=0
MUTE_LOGIC_LINKING=0`
No result at all.

sm0svx, 2021-02-14:
I guess you mean SvxReflector and not Svxserver, unless you are running some other fork of SvxLink.

I believe Adi is correct. You should set MUTE_LOGIC_LINKING=0 in your ModuleEchoLink configuration. It's not a ReflectorLogic configuration. You can find that documented in at least three places: the ChangeLog, the manual page and the git commit log. I'ts also in the default config file even though it's commented out.

When running SvxLink from git, always take the habit to run "git log HEAD@{1}.." after a "git pull" and you will see what have been changed. At the very least, have a look in the svxlink/ChangeLog file.

dl7ata, 2021-02-14:
Sorry Tobias - of course SvxReflector on Master branch.
It doesn't work. I put MUTE_LOGIC_LINKING=0 into ModuleEchoLink.conf without any result.
In the moment I activate **any** module - even metar - the audio stream into reflector is broken and my colleagues can't  hear me.

dl7ata, 2021-02-14:
Ok, in my first try I put _MUTE_LOGIC_LINKING=0_  at the end of ModueEchoLink.conf, just before DESCRIPTION. It won't work. Now it's in line 3 - everything is fine!
Sorry for confusion and thanks for the hint _"git log HEAD@{1}.." after a "git pull"_ !

closed 2021-02-14 by dl7ata

sm0svx, 2021-02-14:
Ok great that you solved it in the end!

The order of config variables should not matter but it's important to understand that the configuration is per module. That is because most modules, like metarinfo, are of node local interest only. It is annoying for everyone in the reflector network to have to listen to silence and DTMF commands.

The most problematic module probably is the Parrot module since some people think it is fun to activate it during a reflector QSO to sabotage for everyone. But that is unfortunately something we have to live with in ham radio. There will always be people sabotaging for others, for some unknown reason.