sm0svx/svxlink #536: Location Info doesn't work on release 19.09.1

IU7GQZ, 2021-03-04:
Hello,

I think I have configured all the parameters correctly, but Location Info does not work and I cannot see the position of my Node on APRS.fi.

I attach below the configuration of the parameters.

[LocationInfo]
APRS_SERVER_LIST=euro.aprs2.net:14580
STATUS_SERVER_LIST=aprs.echolink.org:5199
LON_POSITION=16.55.24E
LAT_POSITION=40.47.34N
CALLSIGN=EL-IU7GQZ
FREQUENCY=144.800
TX_POWER=10
ANTENNA_GAIN=6
ANTENNA_HEIGHT=20m
ANTENNA_DIR=-1
PATH=WIDE1-1,WIDE2-2
BEACON_INTERVALL=10
COMMENT=EchoLink svx

what am I doing wrong?

Tnx

Pino, IU7GQZ

dl1hrc, 2021-03-04:
Ciao Pino,

have you activated LOCATION_INFO in the [GLOBAL] section of svxlink.conf:

[GLOBAL]
...
LOCATION_INFO=LocationInfo
...

Tanti saluti
Adi / DL1HRC

f5vmr, 2021-03-04:
I was going to add the typo error in INTERVALL - should be INTERVAL

Chris

IU7GQZ, 2021-03-04:
> Ciao Pino,
> 
> have you activated LOCATION_INFO in the [GLOBAL] section of svxlink.conf:
> 
> [GLOBAL]
> ...
> LOCATION_INFO=LocationInfo
> ...
> 
> Tanti saluti
> Adi / DL1HRC

Ciao Adi,

Section LOCATION_INFO in activated.

when I start SVXLink with the Sudo svxlink command, the system, on the screen, shows me the correct connection with the APRS server.

below the string I see on the screen:

Connected to APRS server 62.77.224.245 on port 14580

IU7GQZ, 2021-03-04:
> I was going to add the typo error in INTERVALL - should be INTERVAL Chris
> […](#)

Hi Cris

it's just a typo, but in the Svxlink.conf file I wrote the correct command

BEACON_INTERVAL=10

dl1hrc, 2021-03-04:
Ciao Pino,
very strange, really have no idea. Can you please register at

APRS_SERVER_LIST=Leipzig.aprs2.net:14580
This is the APRS server of a friendly OM. He looks in the log to see what's going on. 
Tanti saluti di Adi / DL1HRC

dl1hrc, 2021-03-11:
got no feedback, guess it's solved?

sm0svx, 2021-06-27:
Closing due to inactivity

closed 2021-06-27 by sm0svx