sm0svx/svxlink #537: Setting speed of serial device

dl1hrc, 2021-03-11:
Hi Tobias,
I suspect a problem with the correct initialization of the serial interfaces, especially in the class Serial, method setParams. I suspect that the speed option has no effect on the interface speed, i.e. the parameter when calling the device-> setParams (460800, ..., ..., ...,) method does not initialize or adjust it accordingly.
I was able to determine this when I started my attempts to communicate with Tetra devices, which generally communicate with 115k2. In had to set up the interface with e.g. minicom first, only then did SvxLink communicate with the device although the correct baud rate was configured accordingly. Now I checked it with strace and found the following:

```
write (1, "Baud rate 460800 \ n", 15) = 15
openat (AT_FDCWD, "/ dev / ttyUSB1", O_RDWR | O_NOCTTY | O_NONBLOCK) = 3
ioctl (3, TCFLSH, TCIOFLUSH) = 0
ioctl (3, TCGETS, {B9600 opost isig icanon echo ...}) = 0
write (3, "\ 0 \ 0 \ 0 \ 0 \ 0 \ 0 \ 0 \ 0 \ 0 \ 0", 10) = 10
```

Although setParams () is called immediately after the open, the baud rate is apparently still treated with 9600.
The "`write (3," \ 0 \ 0 \ 0 \ 0 \ 0 \ 0 \ 0 \ 0 \ 0 \ 0 ", 10) = 10`" takes place at a much later point in time in a method I created in which I use the supply connected device with data (would like to).
BTW: I added the B460800 into my Serial class, but the problem also persists with 115k2

mni 73s de Adi / DL1HRC

sm0svx, 2021-03-11:
I only have one serial device at hand right now so I cannot test if it's really working but using the async/demo/AsyncSerial_demo.cpp demo application I get the following which at least look correct.
```
openat(AT_FDCWD, "/dev/ttyUSB0", O_RDWR|O_NOCTTY|O_NONBLOCK) = 3
ioctl(3, TCFLSH, TCIOFLUSH)             = 0
ioctl(3, TCGETS, {B9600 -opost -isig -icanon -echo ...}) = 0
ioctl(3, SNDCTL_TMR_START or TCSETS, {B460800 -opost -isig -icanon -echo ...}) = 0
ioctl(3, TCGETS, {B460800 -opost -isig -icanon -echo ...}) = 0
ioctl(3, SNDCTL_TMR_CONTINUE or TCSETSF, {B460800 -opost -isig -icanon -echo ...}) = 0
```
I temporarily added the B460800 and modified the speed and device in the demo application. Can you do the same test and make it fail?

dl1hrc, 2021-03-12:
Ok, I found the "problem". The parameters may only be set after the device has been opened successfully, not before.
BTW: Is it possible to add 460k8 in AsyncSerial-class?
```
     case 460800:
       serial_speed = B460800;
       break;
```
It is required for the DV3k stick.
mni 73s de Adi / DL1HRC

closed 2021-03-12 by dl1hrc