sm0svx/svxlink #541: GPIO Setup Script - Incorrect Path to gpio number

JGPorteous, 2021-04-04:
Good day,
I trust you are well.

I noticed an issue in the GPIO setup script, hope this is not by design ;)

I was getting the following error:
root@NanoPi-NEO:/sys/class/gpio# /usr/sbin/svxlink_gpio_up
sh: echo: I/O error
*** ERROR: Failed to export GPIO pin "66" (pin number 66)

I added extra debugging/echos to the script and saw that the path looked incorrect:
root@NanoPi-NEO:/sys/class/gpio# /usr/sbin/svxlink_gpio_up
echo 66 > /sys/class/gpio/export
**/sys/class/gpio/66**
sh: echo: I/O error
*** ERROR: Failed to export GPIO pin "66" (pin number 66)

**The Fix/Workaround**
File: svxlink/src/svxlink/svxlink/svxlink_gpio_up.in

line 63:
local pin_path="${GPIO_PATH}/${PIN}"

changed to:
local pin_path="${GPIO_PATH}/**gpio**${PIN}"

Warm regards,
J

sm0svx, 2021-04-04:
You should use the full pin name in the configuration file, like "gpio66". It's the same convention in the svxlink configuration file. From the manual page:
```
GPIO_SQL_PIN
  If  SQL_DET is set to GPIO this configuration variable is used to choose which GPIO pin to use for squelch input.
  The most common name is gpio<number>, like gpio4. Some GPIO drivers use more complex names, like
  gpio33_pe11. If inverted operation is desired, prefix the pin name with an exclamation mark (!).

  Example: GPIO_SQL_PIN=!gpio4
```
The inverted operation does not apply to the gpio.conf file though.

closed 2021-04-04 by sm0svx

JGPorteous, 2021-04-04:
Thank you Tobias

I had it correct in svxlink.conf though my mistake was in /etc/svxlink/gpio.conf where I was using "XX" instead of "gpioXX".