sm0svx/svxlink #543: Qtel "Could not initialize network listen ports" on start

aer0usa, 2021-04-07:
This seems to be related to [Issue #422](https://github.com/sm0svx/svxlink/issues/422), which is closed but doesn't have a solution posted.

When I try to start Qtel in the terminal, the following errors show:

    bind: Cannot assign requested address
    bind: Cannot assign requested address
    Could not initialize network listen ports

...and Qtel does not launch.

This is on KDE neon 5.21.4, kernel 5.4.0-70-generic x86_64. However, this has been the case for quite a while (years?). I'm just now getting around to asking about it.

Thank you, and please let me know if I can provide any more information!
Eric R

aer0usa, 2021-04-07:
I seem to have solved my own problem,  so I'll document it here in case it could help someone else.

I backed up the conf file: `~/.config/SvxLink/Qtel.conf` to another directory and then deleted Qtel.conf.

I launched Qtel, and it launched with a new Settings dialog. I logged in with my callsign and password and clicked Ok. Then I quit Qtel.

I opened the new Qtel.conf and the backup I made, and copied my bookmarks and location to the new Qtel.conf. Now I'm back in (fine) business.

The old Qtel.conf had a `BindAddress` setting, and the new Qtel.conf does not. So I didn't copy it to the new conf. That seems to have been the problem.

Hope that helps!

closed 2021-04-07 by aer0usa

sm0svx, 2021-06-27:
Yes, thanks for posting a follow-up. I just pushed a fix for this: f1cdc753. It's not perfect but now at least Qtel will start and you have the option of bringing up the settings dialog to fix the problem. It's still a bit confusing since the Bind address field will be cleared automatically so all that have to be done to fix the problem is to save the settings.