sm0svx/svxlink #546: devcal synthesizes wrong frequency for tx calibration

DF2HS, 2021-04-29:
I usually use the function "devcal" for TX calibration as described on [this](https://forum.funk-telegramm.de/thread/666-abgleich-von-svxlink-systemen/) page here.

So I type
`devcal -t -f 1000 -m 2500 -d 2500 /etc/svxlink/svxlink.conf Tx1
`
With the current svx version I cannot use this function anymore. I do not get a 1000Hz sinus wave form out of my soundcard.

I get this instead:
[image]

[image]

When I use the speaker-test function everything is fine:
`speaker-test -t sine -f 1000
`

[image]

Can you please fix this. The last picture prooves that this is not a problem of my soundcard.

Thanks alot!

73
Sandy DF2HS

DF2HS, 2022-09-14:
Can you please fix this, the error still occurs.

73
Sandy DF2HS

sm0svx, 2022-09-15:
Hi Sandy,

I cannot reproduce this problem. Are you sure that you are not overdriving the audio chain?

I would not recommend using the maximum deviation (-d 2500 -m 2500) to do the calibration. The reason is that you are very likely to hit some other limit in SvxLink (e.g. if LIMIT_THRESH is set) or the transmitter (anti-splatter filter). As a rule you should always use a calibration deviation (-d) that is lower than the maximum deviation (-m) to not risking overdriving the audio chain during calibration.

My advice is to not use some random instruction found on the Internet but rather use the manual page (man devcal) on your installed system. If you follow that instruction the calibration should work.

Also make sure that you are not using a patched version of SvxLink that may have introduced changes in the audio path that distort the audio.

sm3sgp, 2025-03-18:
@DF2HS Still a problem?

closed 2025-03-18 by sm3sgp