sm0svx/svxlink #548: fails to run

johnfa155, 2021-05-23:
Hi, I accidentally set the server1 name to server4 of course that won't work but if I try the edit it back to server1 it just closes and doesn't change. I have even tried uninstalling all and deleting all qtel files then reinstalling but I can't get it back to server1

johnfa155, 2021-05-24:
I eventually found the file to alter, its the Qtel.conf file I've now made the change and it works fine

closed 2021-05-24 by johnfa155

sm0svx, 2021-06-27:
I don't know why you could not change this in Qtel. I just tried to set a dummy server name, exit, and then start Qtel again and I could change it back to a real server name without problems.

For anyone else finding this thread, the exact location of the file where Qtel save its settings on most systems is ~/.config/SvxLink/Qtel.conf.

johnfa155, 2021-06-27:
Problems sorted thanks I could only say what was happening on mine, in the end I found the file that holds this info and changed it manually then restarted alls been fine since