sm0svx/svxlink #551: Hick-up with REMOTETRX

dl7ata, 2021-06-26:
Setup:
Main svxlink on BananaPi, remotetrx (HFTx/HRRx) in the same LAN on an Intel NUC running Ubuntu. 
During a QSO a disconnect appears:

`25.06.2021 09:14:34: HFTx: Turning the transmitter ON                                                                                                                           

25.06.2021 09:14:41: Logic_DL7ATA: UDP frame(s) lost. Expected seq=17429 but received 17560. Resetting next expected sequence number to 17561                                    
25.06.2021 09:14:42: *** ERROR: TCP transmit buffer overflow. Disconnecting from 192.168.0.19:5230...
25.06.2021 09:14:42: HFRx: Disconnected from remote receiver 192.168.0.19:5230: Locally ordered disconnect                                                                     
25.06.2021 09:14:42: HFTx: Disconnected from remote transmitter at 192.168.0.19:5230: Locally ordered disconnect                                                               
25.06.2021 09:15:02: 192.168.10.199:5230: RemoteTrx protocol version 2.8                                                                                                         
25.06.2021 09:15:02: HFRx: Connected to remote receiver at 192.168.0.19:5230                                                                                                   
25.06.2021 09:15:02: HFRx: Requesting CODEC "RAW"                                                                                                                                
25.06.2021 09:15:02: HFTx: Connected to remote transmitter at 192.168.0.19:5230                                                                                                
25.06.2021 09:15:02: HFTx: Requesting CODEC "RAW"`

It looks like that the resetting of the UDP frame losses caused a TCP error between main instance and remotetrx.

sm0svx, 2021-06-27:
It's highly unlikely that this is a bug. You probably just had a temporary network problem/blockage so you lost a reflector audio packet and then got disconnected from the RemoteTrx due to the network buffers in the application filling up.

closed 2021-06-27 by sm0svx

dl7ata, 2021-06-27:
Hi Tobias,you are right, I have had a netwok problem with the internet line from svxlink into the svxreflector-server/master. But not between sxlink and remotetrx, this is running in a stable LAN.That means, if UDP-packets between main instance of svxlink and svxreflector-server struggle, also the LAN connection will close?Frank

sm0svx, 2021-06-27:
The error message you get indicate that SvxLink is sending TCP packages to the RemoteTrx but they are not transmitted by the networking layer or not received by RemoteTrx so packets stack up in the buffer and in the end it will cause a buffer overflow. I still think it's highly unlikely that this problem is caused by a bug in SvxLink. Maybe your LAN is not as stable as you think. Networking problems some times are very subtle and hard to find. Try asking at goups.io if anyone else have seen the same problem.

dl7ata, 2021-06-27:
Thanks for your reply and explanation, I will ask the group.Best from BerlinFrank