sm0svx/svxlink #552: Async when using 2 logic sip + simplex logic

sa2blv, 2021-07-12:
svxlink: /home/peter/svxlink/src/async/audio/AsyncAudioSelector.cpp:458: void Async::AudioSelector::selectBranch(Async::AudioSelector::Branch*): Assertion `(m_selected_branch == 0) || (m_selected_branch->streamState() == STATE_IDLE) || (m_selected_branch->streamState() == STATE_WRITING)' failed.
Aborted

can you check sm0svx
// Sa2blv Peter

sa2blv, 2021-07-12:
https://groups.io/g/svxlink/attachment/1009/0/svxlink1-6021.dat

Valgrind dump file

sm0svx, 2021-08-13:
Please provide a way to reproduce this assertion in the svxlink master branch. If I can't reproduce it it's very hard to fix it. Also it could just as well be some other code causing the assertion since you got the assertion in a patched code base.