sm0svx/svxlink #553: Orange PI Zero and gpio

gulweb, 2021-08-11:
Hi,

I fail to get latest svxlink running. The problem is I get no PTT activation. My installation is correct in that I can get the squelch to open, can transmit *# to get an announcement, and svxlink says it is turning transmitter ON and OFF, but my gpio pin does not activate TX. If I manually activate it, by "gpio write 7 1" the tx goes on. I can then do "aplay something.wav" and it is transmitted.

Changing PTT_PIN to something different than gpio7 causes svxlink to fail during start, so this is obviously the only correct value
based on my gpio config. The /sys/class/gpio/gpio7 is there. I have seen a lot written about OpiZ and gpio, but cannot figure out exactly where the problem is. This is something new, used to work before...

uname -a
Linux orangepizero 5.10.43-sunxi #21.05.6 SMP Mon Jun 21 15:07:45 UTC 2021 armv7l GNU/Linux
SvxLink v1.7.99.52

f5nlg, 2021-08-11:
you must define them in /etc/rc.local file to have them active on svxlink

sorry no samples i'm on my phone

73"

f5vmr, 2021-08-11:
Paste the following before the end of the file .fi

# Radio 1 GPIO 18:PTT - GPIO 17:SQUELCH
echo "17" > /sys/class/gpio/export &
sleep 2
echo in  > /sys/class/gpio/gpio17/direction

# Squelch
echo "18" > /sys/class/gpio/export &
sleep 2
echo out > /sys/class/gpio/gpio18/direction

This is an example from one of my repeaters. 
Just adjust your GPIO pins to suit. Then save  /etc/rc.local

Then reboot

73.  Chris

gulweb, 2021-08-11:
This is what I have in rc.local

# SVXLink TX1 PTT gpio7 (pin 12)
echo 7 > /sys/class/gpio/export
echo 'out' > /sys/class/gpio/gpio7/direction
sudo chmod 777 /sys/class/gpio/gpio7/value

this creates the pin 7 and I can toggle it with gpio, and it DOES start the TX. 

gpio readall looks OK.

f5vmr, 2021-08-11:
Excellent - Hope that solves the problem.

73
Chris

f5nlg, 2021-08-11:
for orange pi zero :

echo "10" > /sys/class/gpio/export
echo "in"  > /sys/class/gpio/gpio10/direction

sleep 1

echo "7" > /sys/class/gpio/export
echo "out" > /sys/class/gpio/gpio7/direction
sleep 2

don't forget double quotes

sm0svx, 2021-08-11:
Why not configure GPIO in /etc/svxlink/gpio.conf and use the svxlink_gpio_setup systemd service?

gulweb, 2021-08-11:
The header is wrong in my previous post.
I need to toggle pin 7, which I can do with gpio write 7 0 or 1
However, it looks as this is actually GPIO6 (??) so I'll try to modify this.

sm0svx, 2021-08-13:
Not a bug. See [this thread](https://groups.io/g/svxlink/topic/orange_pi_zero_pin_confusion/84836374) and [this thread](https://groups.io/g/svxlink/topic/still_no_luck_with_orange_pi/84844165) and [this post](https://groups.io/g/svxlink/message/1055) for the resolution. Closing issue.

closed 2021-08-13 by sm0svx