sm0svx/svxlink #554: Manual DTMF input via keyboard console notes and bugs.

hayden-t, 2021-08-15:
svxlink 1.7.0
Hi, ive been trying out the ability to input dtmf commands as mentioned here via the keyboard.
https://github.com/sm0svx/svxlink/wiki/InstallationInstructions#svxlink-server-configuration

- First this only works if you are not using the --logfile option, this may be a bug or unavoidable.
- While other DTMF commands work, '*' (self identify) is not working via this method.
- The injected DTMF command is hard coded to go to the first listed logic in svlink.conf [GLOBAL] → LOGICS, in an ideal world one could say press TAB to cycle between the different logics that codes would be sent to.

f5vmr, 2021-08-15:
That is correct. Once the daemon is running, or you are running the svxlink.service, the DTMF tones have to be by RF. 

For the SVXReflector there is a special case to activate the TalkGroup scenario, unlike the activation of EchoLink.

[LinkToReflector]
CONNECT_LOGICS=SimplexLogic:9:,ReflectorLogic
DEFAULT_ACTIVE=1
OPTIONS=DEFAULT_CONNECT,NO_DISCONNECT
TIMEOUT=0

Unless you want to be disconnected when other modules are activated whilst a connection is in play you have to add the line
MUTE_MODULE_LINKING=0 in the ModuleEchoLink.conf oor the ModuleMetarInfo.conf.

If you want the interruptions in this case then make the Boolean choice of =1.

So a Talkgroup Sequence will be 91’235’# as an example for talk group 235, 92’2350’# as QSY to 2350. And 9# for disconnect the svxreflector.
I have it running on my svwreflector server in this fashion.

Chris G4NAB.

hayden-t, 2021-08-15:
For this testing purposes with the keyboard im neither running in daemon mode or service

hayden-t, 2021-08-15:
So I also found out there is another way to input DTMF via keyboard and that is via DTMF_CTRL_PTY  setting which lets you do it while svxlink is in the background eg 

for example use "echo" to execute commands:

echo -n "*#" >/dev/shm/repeater_dtmf_ctrl

Or if you want to type commands like in the console you can use the 'socat' utility:

socat FD:0,icanon=0,echo=0 OPEN:/dev/shm/repeater_dtmf_ctrl