sm0svx/svxlink #555: DTMF_MUTING detection delay time, manual incorrect, bonus make it adjustable.

hayden-t, 2021-08-15:
In http://www.svxlink.org/doc/man/man5/svxlink.conf.5.html it states 75ms detection delay, but that is not correct afaik. 
Default decoder type INTERNAL only has 40ms delay
https://github.com/sm0svx/svxlink/blob/4ee98eb7e2a9bd3c1b7c3667b84b704cb2bc9925/src/svxlink/trx/SvxSwDtmfDecoder.h#L165
DH1DM has 75ms

But i found in my situation even that is not enough, as the start of code still coming through, ideally this parameter could be exposed to the conf file.

hayden-t, 2021-08-15:
[image]

f5vmr, 2021-08-16:
Since using the ReflectorLogic, I've have never encountered any difficulty
with DTMF either in sending or muting, provided DEEMPHASIS=0 is set. With
this parameter set to 1 some tones fail, in my case the 3 from my Yaesu
FT60. Naturally the filtering instigates a delay when RepeaterLogic is
running and users will detect their voice on repeater systems because of
the delay.

Chris

sm3sgp, 2025-03-18:
@hayden-t The man page seams to be updated with a new text saying 40-75ms. I close this issue.

closed 2025-03-18 by sm3sgp