sm0svx/svxlink #556: playFile relative path does not seem to work, only absolute ?

hayden-t, 2021-08-16:
[image]
https://github.com/sm0svx/svxlink/wiki/InstallationInstructions#the-event-handling-subsystem

f5vmr, 2021-08-16:
In practice relative path works fine. If the DEFULT_LANG is set correctly
then I've never had any problem with playing a file set by play. playFile
on the other hand I've never used, never setting the parameter in Logic.tcl.
Chris

hayden-t, 2021-08-16:
where is play method ? i do not see it

f5vmr, 2021-08-16:
The playSound and other commands are found in the .tcl logic files. Unless
you know what your doing and copy the files to a new directory called local
to protect the originals, you risk damage. The files normally reside at
/usr/share/svxlink/events.d but creating a directory called local here
gives copies you can work on, that SVXLink takes in preference when they
are edited. Warning - make changes at your own risk. The play commands come
in many forms, play, playSound, CW::play and spell commands too.

Chris

f5vmr, 2021-08-16:
The only file you should look at is Logic.tcl being attentive to syntax.

Chris

hayden-t, 2021-08-16:
I know all about this. you are not talking about the functions im talking about in my issue.