sm0svx/svxlink #557: web man5/svxlink.conf.5.html is out of date with git version

hayden-t, 2021-08-16:
http://www.svxlink.org/doc/man/man5/svxlink.conf.5.html

https://github.com/sm0svx/svxlink/blob/master/src/doc/man/svxlink.conf.5

Im finding many new features I did not know about from the git version which isnt easy to read. Though can also be read via terminal with command:  man svxlink.conf

sm3sgp, 2021-08-16:
The web page corresponds to the latest official released of svxlink. (unfortunately a bit old).
Always use the installed man page for the version you are using, using man svxlink.conf on your system.

hayden-t, 2021-08-16:
How do i find out what the latest official build is to compare / check this ?

f5vmr, 2021-08-16:
The master is the latest build. Although it is version 19.1 something, this
doesn't mean that that it hasn't been updated. You can see the dates of the
changes to various modules along side the files in the GitHub. There was
one update yesterday.
The only way to get the 'latest' latest version is to go to your current
install folder and sudo git pull. That will introduce the updated files.
Then sudo make, sudo make doc, then sudo make install, then finally sudo
ldconfig.
You need not do a sudo cmake, as this will be done automatically by the
existing framework if the need arises.

It shouldn't take as long as a full compile but will overwrite the newer
files, except the .conf files. If you have made a local folder for the .tcl
files, these will not be overwritten, so if there is code you have changed,
you will need to redo these changes in the new .tcl files in the events.d
folder.

I'm sorry if this doesn't answer your question but the only way to see the
up-to-date man files is to perform the sudo make doc that builds the new
man files for you.

Chris

sm3sgp, 2021-08-17:
The latest release  is 19.09.2. It can be found on this page:
https://github.com/sm0svx/svxlink/releases

It will be difficult to keep the documentation links up to date with different version. As mentioned, the best is to use the man pages an the terminal: man svxlink.conf

Since it is not a bug in the software I close this issue for now. Thanks for bringing this up.

closed 2021-08-17 by sm3sgp

hayden-t, 2021-08-17:
my build reports as SvxLink v1.7.0 ??? how can this be compared ?? as i feel its not that new but still had features in the man not on the website....

sm3sgp, 2021-08-17:
If you compiled from source, check this file:
https://github.com/sm0svx/svxlink/blob/master/src/versions

hayden-t, 2021-08-17:
im not compiling from source currently. raspberry pi in ics 2x system

hayden-t, 2021-08-17:
I inquired with by system provider and i have 17.12.2 so what im saying is that lined man file is not even up to date with that. I can understand it not being bleeding edge, but 17.12 is over 3 years old ?!?!