sm0svx/svxlink #558: svxlink add --version option that reports build release rather than v1.7.9

hayden-t, 2021-08-17:
Hi currently when svxlink starts it reports v1.7.9 which doesnt seem to correlate to anything.

It would be ideal if there was a -v option for the binary that reports the git code version from compile time.

dl7ata, 2021-09-05:
Check [https://github.com/sm0svx/svxlink/blob/master/src/versions](url).
You'll find any version of any peace of software using in svxlink.

Please don't flood the issue tracker and use [https://groups.io/g/svxlink/topics](url) for your requests

hayden-t, 2021-09-05:
not sure this is what im looking for, but your link href is wrong

dl7ata, 2021-09-05:
Don't know why it doesn't work. Find out by yourself at https://github.com/sm0svx/svxlink/blob/master/src/versions

hayden-t, 2021-09-05:
thats better, the other ones underlying link is different. so that is nice to know about but its not what this issue about. that will only tell you what the current latest version is, but i want to know what a builds version is

f5vmr, 2021-09-05:
Good day - A git pull of the master will do everything you need….. It’s what I use currently on SVXPortal (svxportal-uk.ddns.net <http://svxportal-uk.ddns.net/>)

If you need the USRP version for DVSwitch then you’ll have to go to the dl1hrc GitHub.

Regards

Chris

hayden-t, 2021-09-05:
guys, im not talking about git source but compiled packages

f5vmr, 2021-09-05:
That the problem - I compile from scratch from the GitHub. The OpenRepeater Packages are behind, and don’t contain the SVXReflector options.

If you need an instruction sheet I can help you. A full build on a Raspberry Pi - quad core, takes about 45-60 minutes.
 
Chris

hayden-t, 2021-09-06:
to help me understand this better, so i see in the startup of the app it prints SVXLINK_VERSION as the version but i cant find where it is set ?

closed 2021-09-06 by hayden-t