sm0svx/svxlink #560: Ubuntu in Virtual Box.. Qtel does not connect

Bianchicycle, 2021-09-01:
[image]
[image]

I am running on a MAC and using Ubuntu Bungie (newest vs) inside of a Virtual Box. When I start Qtel, the stations propagate fine which say I AM connecting to the main server. When I try to connect to a link...NOTHING. It just disconnects after a while. I have attached the server page and the sound page. The port ARE open! I have read the article "Sound problem with Qtel )open source Echolink) in Ubuntu 14.04. HELP!!!

dj1an, 2021-09-02:
Is your VM Guest Net bridged to your Host Network Adapter?
Else Virtualbox is providing a guest NAT, which would prevent your port Forwarding.

Bianchicycle, 2021-09-02:
I would think so if I can browse the internet vis the build-in (Ubuntu menu) browser. Would this be true?

dj1an, 2021-09-02:
Please check your Guest Machine Settings if it is configured as a bridged Adapter:

[image]

Bianchicycle, 2021-09-02:
These are the only network controls I can find.