sm0svx/svxlink #571: svxlink connection problem on raspberry pi

ibodumaan, 2022-01-23:
Hello,
My system hasn't been working for about 3 months. There was nothing at first, but it started giving a timeout error even though I didn't update at all. I reinstalled everything but the result is the same. Internet connection is ok, it seems active, but it gives timeout error all the time. UDP-(5198 and 5199), TCP(5200) ports are open. Do I need to open another port?

I would be glad if you help.

TA6IBD

dl7ata, 2022-01-29:
This is probably not an issue of svxlink. Please use to identify your problem [https://groups.io/g/svxlink](url) and report to the group exactly you problem, log and some more detailed information.

Luigiiz0yay, 2022-03-15:
I think that i have same problem. Yhe Echolink module is connect, but after 120 seconds any QSO is Muted: my node does not transmit. If the node recive by RF a call or a tone DTMF it restart to TX for only 120 Second