sm0svx/svxlink #574: ERROR: Transmitter Tx1 have been active for too long.

DL2DTM, 2022-02-15:
Hallo Tobias,

i get the above error message at least once a week. Otherwise the RASPI runs with svxlink and reflector without any problems.
Is it possible to end the program with an error after the message has appeared, in order to then trigger a restart from the systemd?
My log is attached.
[svxlink_error.txt](https://github.com/sm0svx/svxlink/files/8066703/svxlink_error.txt)

DG1FI, 2022-06-15:
Hi Karl, please provide your [RepeaterLogic] and [TX1] Section.

73 Mario

DL2DTM, 2022-06-20:
Hallo Mario,
 
vielen Dank für deine Info. Als Anlage sende ich Dir die Dateien zu.
 
73 von DL2DTM
Karl-heinz

too long. (Issue #574)
Datum: 2022-06-15T18:47:53+0200
 
 
 
 

<https://github.com/sm0svx/svxlink/issues/574#issuecomment-1156705903> , or 
unsubscribe
[ { ***@***.***": "http://schema.org", ***@***.***": "EmailMessage", 
"https://github.com/sm0svx/svxlink/issues/574#issuecomment-1156705903", 
"url": 
"https://github.com/sm0svx/svxlink/issues/574#issuecomment-1156705903", 
"name": "View Issue" }, "description": "View this Issue on GitHub", 
"https://github.com" } } ]

sm3sgp, 2025-01-14:
@DL2DTM Any update on this?

DL2DTM, 2025-01-20:
Hello Gunnar,thank for your request. I had change my RASPI and update svxlink software. My system works well!73 von DL2DTM  Karl-Heinz Von meinem/meiner Galaxy gesendet

closed 2025-01-20 by sm3sgp