sm0svx/svxlink #577: Compiler error: Assertion `isIdle() && con->isIdle()' failed

dl7ata, 2022-04-14:
When compiling latest Version 1.7.99.66 opening svxlink runs into an error:
svxlink: /home/svxlink/svxlink/src/async/core/AsyncTcpClientBase.cpp:229: void Async::TcpClientBase::connect(): Assertion `isIdle() && con->isIdle()' failed.

sm0svx, 2022-04-14:
ok is that a pure unmodified SvxLink master från my git repo or with modifications from another fork?  In any case we need a stack trace for that so run it under gdb to get a stack trace, something like:
```
$ gdb svxlink
(gdb) run
<assertion failed>
(gdb) bt
```
NOTE: There have been a substantial rewrite of the networking code so some instabilities are to be expected. The assertion probably comes from calling "connect" on an already connected socket, which is not allowed anymore.

dl7ata, 2022-04-14:
Thanks for your quick reply. gdb didn't run to the assertion failure, see below.

I assume that there could be a problem with the remote-trx. When I deactivate my simplex-remote section in the config, it runs without any error. The remote-trx is currently not online, so svxlink could not find the client in LAN when starting.

The gdb-out:

> (gdb) run
Starting program: /usr/bin/svxlink 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".

Program received signal SIGILL, Illegal instruction.
0xb62f0728 in ?? () from /usr/lib/arm-linux-gnueabihf/libcrypto.so.1.1
> (gdb) bt
#0  0xb62f0728 in ?? () from /usr/lib/arm-linux-gnueabihf/libcrypto.so.1.1
#1  0xb62ea18c in ?? () from /usr/lib/arm-linux-gnueabihf/libcrypto.so.1.1
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb)<

sm0svx, 2022-04-14:
That was strange. It did not start at all. Make sure any other running instance is stopped. Try to start svxlink on the command line without gdb. Illegal instruction is not expected to happen.

I'll test RemoteTrx. I'm not sure that I have verified that is is functional after my networking code changes.

dl7ata, 2022-04-14:
No other instance is running. I started svxlink on the command line:

> ....
Starting logic: Logic_DL7ATA
Logic_DL7ATA: Connecting to service 

> Starting logic: Logic_remote
Loading RX: HFRx
Loading TX: HFTx
svxlink: /home/svxlink/svxlink/src/async/core/AsyncTcpClientBase.cpp:229: void Async::TcpClientBase::connect(): Assertion `isIdle() && con->isIdle()' failed.
Abgebrochen

sm0svx, 2022-04-14:
Yep there seem to be a problem when connecting to the RemoteTrx application. I'll have a look at it.

dl7ata, 2022-04-14:
This is the last section in svxlink.conf which is loaded:

> [HFRx]
TYPE=Net
RX_ID=3
HOST=regulus
TCP_PORT=5230
LOG_DISCONNECTS_ONCE=1
AUTH_KEY="svxlink_DB0TGO"
CODEC=RAW

> [HFTx]
TYPE=Net
HOST=regulus
TCP_PORT=5230
LOG_DISCONNECTS_ONCE=1
AUTH_KEY="svxlink_DB0TGO"
CODEC=RAW

sm0svx, 2022-04-14:
New code pushed. Try it now.

dl7ata, 2022-04-14:
Bug fixed - system is running! _[I didn't checked the remote system because there is no access at the moment]_
Thanks for quick response, thanks for your work at all!!

closed 2022-04-14 by dl7ata