sm0svx/svxlink #581: invalid command name "say_talkgroup" in V1.7.99.69

dl7ata, 2022-05-23:
After updating to 1.7.99.69, I got this error in the log file:
*** ERROR: Unable to handle event: ReflectorLogic::tg_command_activation 2621 0 in logic ReflectorLogic (invalid command name "say_talkgroup")

sp2ong, 2022-05-23:
I suppose you need to update your ReflectorLogic.tcl file to the latest version

https://github.com/sm0svx/svxlink/blob/master/src/svxlink/svxlink/ReflectorLogic.tcl#L50

dl7ata, 2022-05-23:
I did.

f5vmr, 2022-05-23:
Please ensure that in the [Reflector logic] paragraph that there is a line
DEFAULT_LANG= your language. This was a puzzle to me when I first used
protocol 2.
Chris

f5vmr, 2022-05-23:
Then check that the sound sample is of the correct type. The 12th byte
should be Hex 10 and not Hex 12. Otherwise svxlink may reject it. I use
en_GB of my own construction and everything works as it should.

Chris

dl7ata, 2022-05-23:
Chris, _DEFAULT_LANG=de_DE_ or _DEFAULT_LANG=us_US_ doesn't work. Now I got the announcement in German, but I am missing the number of the talkgroup. So same behavior with and without the parameter.
The soundfiles are working for some years, I didn't change anything.

sp2ong, 2022-05-23:
I suppose you don't use your own ReflectorLogic.tcl in /usr/share/svxlink/event.d/local/
Sometimes some people make their own modified copy in this directory and "make install"
 just rewrites to new version only in  /usr/share/svxlink/event.d/

f5vmr, 2022-05-23:
Yes that's possible. I too use all logics in a 'local' folder. If a pull is
made and a further make, the local files do not get overwritten.

Best of luck finding the solution

dl7ata, 2022-05-23:
Of course I use my own logic and compared it with the one in  /usr/share/svxlink/event.d/ .
The _proc tg_command_activation _ is exactly the same. And in V 1.7.99.68 everything run smooth.
I'll let you know when I have found a solution - thanks for your help anyway!

dl7ata, 2022-05-23:
Shame on me - I got it. I missed the new procedure _proc say_talkgroup_ in my own own tcl.
Thanks again gentlemen!

closed 2022-05-23 by dl7ata