sm0svx/svxlink #589: ReflectorLogic, no reconnect after IP failure

dl8bby, 2022-07-02:
With recent versions of the client software (e.g. v1.7.99.68/69), there is no reconnection to the reflector server after an IP failure, if the IP address is entered for HOSTS instead of the hostname. Typical entry in the log file:

24.06.2022 18:00:08: ReflectorLogic: Heartbeat timeout
24.06.2022 18:00:08: ReflectorLogic: Disconnected from 192.168.1.55:5300: Locally ordered disconnect
24.06.2022 18:01:08: ReflectorLogic: Connecting to service
24.06.2022 18:01:08: ReflectorLogic: Disconnected from 192.168.1.55:5300: Network is unreachable
24.06.2022 18:01:09: ReflectorLogic: Disconnected from 192.168.1.55:5300: Network is unreachable

After that, no further connection attemps (Connecting to ..., Disconnected ...)

After the return of the IP connection, logging into the Echolink server and APRS is no problem, but there are no further messages from the ReflectorLogic and the connection to the server is not reestablished.

The problem has already been described on groups.io by IK4NZD. So far, there is no viable solution. The tip "use the hostname für HOST instead" is not feasible in every network.

73 de DL8BBY, Matthias

sm0svx, 2022-07-03:
I cannot replicate the problem when setting HOSTS to one IP address. The node still reconnect after a network failure.
Try to activate some more debugging printouts by uncommenting #define ASYNC_STATE_MACHINE_DEBUG and recompile.
https://github.com/sm0svx/svxlink/blob/master/src/async/core/AsyncTcpPrioClientBase.cpp#L53

It would help a lot if you could find a way, a step by step instruction, that I can use here to replicate the problem.

sm0svx, 2022-08-04:
Fixed in SvxLink 1.7.99.71

commit 7ebd9ed5 (2022-08-04): Improve Async::StateMachine stability

Fixes #589 issue with ReflectorLogic, no reconnect after IP failure.
closed 2022-08-04 by sm0svx