sm0svx/svxlink #591: invalid command name "CW::setAmplitude"

booyaamps, 2022-08-05:
Hi! Im moving our SVXlink controller to a new machine. i've compiled it and brought all of the logics over from the old machine, and whenever i try to start svxlink, i get this error:
/usr/share/svxlink/events.tcl in logic RepeaterLogic: invalid command name "CW::setAmplitude"
Svxlink does not start.
Any ideas as to what i'm missing?

Dloranger, 2022-08-05:
likely you need to refresh your logics with the new ones that have these
functions provided

booyaamps, 2022-08-05:
Im restoring from a nightly tarball that we have backing up /etc/svxlink
and /usr/share/svxlink, and all of the logics are working properly on the
existing(older pc) system. We are just trying to migrate it over to a newer
pc, with the same build version of svxlink being used. The logics should be
the same versions on both pcs.

closed 2022-08-07 by booyaamps

Dloranger, 2022-10-11:
are those backups inclusive of the subdirectories respectively such as
events.d/ etc?

Seems like there might be a modification in the TCL files on the old
machine that is not being brought along to the new machine

in the repeaterlogic.tcl
<https://github.com/sm0svx/svxlink/blob/master/src/svxlink/svxlink/RepeaterLogic.tcl>
source
I dont see anything related to the CW functions, nor in the events.tcl
<https://github.com/sm0svx/svxlink/blob/master/src/svxlink/svxlink/events.tcl>
files so perhaps one of those has been modified from the defaults to add
the new functionality?

Just guessing, but I would check to make sure everything you intended/need
is being brought along as a first step