sm0svx/svxlink #593: Problem at launch 1_7_99_72

ik4nzd, 2022-08-20:
I installed the new version 1_7_99_72, but at launch this error comes out, which seems to be related to the events.tcl file.
I can't find where the missing parenthesis should be put .....
In previous updates the problem had never occurred .....
Thanks in advance for the help.
'73 de Adam

Sat Aug 20 19:25:04 2022: Loading module "ModuleMetarInfo" into logic "SimplexLogic"
Sat Aug 20 19:25:04 2022: 	Found /usr/lib/arm-linux-gnueabihf/svxlink/ModuleMetarInfo.so
Sat Aug 20 19:25:04 2022: 	Module MetarInfo v1.2.1.99.4 starting...
Sat Aug 20 19:25:04 2022: **/usr/share/svxlink/events.tcl in logic SimplexLogic: missing close-brace: possible unbalanced brace in comment**
Sat Aug 20 19:25:04 2022: *** ERROR: Could not initialize Logic object "SimplexLogic". Skipping...
Sat Aug 20 19:25:04 2022: 
Sat Aug 20 19:25:04 2022: Starting logic: ReflectorLogic
Sat Aug 20 19:25:04 2022: **/usr/share/svxlink/events.tcl in logic ReflectorLogic: missing close-brace: possible unbalanced brace in comment**
Sat Aug 20 19:25:04 2022: *** ERROR: Could not initialize Logic object "ReflectorLogic". Skipping...
Sat Aug 20 19:25:04 2022: *** ERROR: No logics available. Bailing out...
Sat Aug 20 19:25:10 2022: SvxLink v1.7.99.72 Copyright (C) 2003-2022 Tobias Blomberg / SM0SVX

sm0svx, 2022-08-20:
I don't get this on any of my systems. The only TCL file that have changed is ReflectorLogic.tcl. This is new TCL code for example:
https://github.com/sm0svx/svxlink/blob/e7e69b0e70baf200a6d6b5b532aa22c6a0edf6df/src/svxlink/svxlink/ReflectorLogic.tcl#L80-L95
I've heard people getting in trouble with commented out TCL code but I have never seen that. You could try to remove the lines starting with a "#" to see if it helps but it should not be necessary.

Maybe you have customized TCL files that cause this?  Try to temporarily remove any customizations to see if the problem persists.

Also note that this commit requires a new sound pack to add the "reflector" sound clip and the "connect" and "disconnect" sound clips have been moved from the EchoLink context to the Default context.

ik4nzd, 2022-08-20:
Hi Tobias and thanks for the reply.
I solved it in the end by recompiling the software.
Then I edited my ReflectorLogic.tcl file with some previous changes and now everything works fine.
Previously, however, he did not want to start either with the original files or with the modified one.
**Since changes have been made to the audio files in the version, is there a changelog of the changes that are made from time to time? It would be very useful to know where to create or move files when there are variations.**
Thank you

ik4nzd, 2022-08-20:
> Also note that this commit requires a new sound pack to add the "reflector" sound clip and the "connect" and "disconnect" sound clips have been moved from the EchoLink context to the Default context.

Maybe I think the files go to the Core directory and not the Default ..... right?

sm0svx, 2022-08-20:
Looking at the latest commit for svxlink-sounds-en_US-heather is what you need to do if you are a langpack maintainer: https://github.com/sm0svx/svxlink-sounds-en_US-heather/commit/1f11f3c9dc4507acd86250a51849fbff42955729. If you use the en_US sound pack you just need to download the latest version of course.

In short:
New file: Core/reflector.wav
EchoLink/connected.wav: Moved to Default/connected.wav
EchoLink/disconnected.wav: Moved to Default/disconnected.wav

The connected/disconnected sound clips need to be in the Default directory for both the Core and the EchoLink module to find them.

closed 2022-08-20 by sm0svx