sm0svx/svxlink #595: ModuleEchoLink. Incorrect message from the directory server.

ur3qjw, 2022-08-30:
Hello Tobias!
I noticed that I began to receive incorrect messages from the directory server.
If the EchoLink password is incorrect, the server, returns the Ok status anyway.
Tested on SvxLink v.1.7.99.67, 1.7.99.72.

Log output. The password is initially incorrect.
30.08.2022 20:01:07: EchoLink directory status changed to ON
30.08.2022 20:01:08: --- EchoLink directory server message: ---
30.08.2022 20:01:08: EchoLink Server v2.6.104
30.08.2022 20:01:08: 
30.08.2022 20:01:08: ECHO6: N.Virginia, USA

sm0svx, 2022-08-30:
Sounds more like a bug in the Echolink directory server then if it lets you in even though you are giving it the incorrect password, doesn't it?

ur3qjw, 2022-08-30:
If the password is incorrect, the station is not in the station list.
Windows client v.2.0.908, also writes: "Incorrect password".

[image]

commit 1a8ea722 (2023-07-19): Bugfix in ModuleEchoLink: INCORRECT PASSWORD

The INCORRECT PASSWORD message was ignored in ModuleEchoLink. It is now
caught and an error is indicated.

Fixes #595
closed 2023-07-19 by sm0svx