sm0svx/svxlink #596: Setting up a simplex parrot server

WriteCodeEveryday, 2022-09-04:
I have a cheap C-Media USB adapter that I've connected to a device trying the following config.

`[GLOBAL]
#MODULE_PATH=/usr/lib/arm-linux-gnueabihf/svxlink
LOGIC_CORE_PATH=/usr/lib/arm-linux-gnueabihf/svxlink
LOGICS=ParrotOverAudioLogic
CFG_DIR=svxlink.d
TIMESTAMP_FORMAT="%c"
CARD_SAMPLE_RATE=48000
#CARD_CHANNELS=1
#LOCATION_INFO=LocationInfo
#LINKS=LinkToR4

[ParrotOverAudioLogic]
TYPE=Repeater
RX=MicInput
TX=HeadphoneOutput
MODULES=ModuleParrot
CALLSIGN=KO4YZK
SHORT_IDENT_INTERVAL=10
LONG_IDENT_INTERVAL=60
EVENT_HANDLER=/usr/share/svxlink/events.tcl
DEFAULT_LANG=en_US
RGR_SOUND_DELAY=0
FX_GAIN_NORMAL=0
FX_GAIN_LOW=-12
IDLE_TIMEOUT=30
IDLE_SOUND_INTERVAL=3000

[MicInput]
TYPE=Local
AUDIO_DEV=alsa:plughw:1
AUDIO_CHANNEL=1
SQL_DET=VOX
VOX_FILTER_DEPTH=100
VOX_THRESH=500
PEAK_METER=1

[HeadphoneOutput]
TYPE=Local
AUDIO_DEV=alsa:plughw:1
AUDIO_CHANNEL=0
PTT_TYPE=NONE
TX_DELAY=500
`

Tried both AUDIO_CHANNEL 0 and AUDIO_CHANNEL 1 but it won't trigger the Parrot to work.

`Sun Sep  4 06:52:05 2022:       Found plugin: /usr/lib/arm-linux-gnueabihf/svxlink/RepeaterLogic.so
Sun Sep  4 06:52:05 2022: ParrotOverAudioLogic: Loading RX "MicInput"
Sun Sep  4 06:52:05 2022: ParrotOverAudioLogic: Loading TX "HeadphoneOutput"
Sun Sep  4 06:52:05 2022: ParrotOverAudioLogic: Loading module "ModuleParrot"
Sun Sep  4 06:52:05 2022:       Found /usr/lib/arm-linux-gnueabihf/svxlink/ModuleParrot.so
Sun Sep  4 06:52:05 2022:       Module Parrot v1.1.1 starting...
Sun Sep  4 06:52:05 2022: ParrotOverAudioLogic: Event handler script successfully loaded.
Sun Sep  4 06:52:35 2022:
Sun Sep  4 06:52:35 2022: SIGTERM received. Shutting down application...
`

Changing the AUDIO_CHANNEL beyond 1 is no good.

` Using configuration file: /etc/svxlink/svxlink.conf
Sun Sep  4 06:50:35 2022: --- Using sample rate 48000Hz
Sun Sep  4 06:50:35 2022:
Sun Sep  4 06:50:35 2022: Starting logic: ParrotOverAudioLogic
Sun Sep  4 06:50:35 2022:       Found plugin: /usr/lib/arm-linux-gnueabihf/svxlink/RepeaterLogic.so
Sun Sep  4 06:50:35 2022: ParrotOverAudioLogic: Loading RX "MicInput"
Sun Sep  4 06:50:35 2022: *** ERROR: Audio channel out of range when opening audio device "plughw:1. The card have 2 channel(s) configured, but (zero based) channel number 2 was requested.
Sun Sep  4 06:50:35 2022: *** ERROR: Could not open audio device for receiver "MicInput"
Sun Sep  4 06:50:35 2022: *** ERROR: Could not initialize RX "MicInput"
Sun Sep  4 06:50:35 2022: *** ERROR: Could not load or initialize Logic object "ParrotOverAudioLogic". Skipping...
Sun Sep  4 06:50:35 2022: *** ERROR: No logics available. Bailing out...
`

I don't know how I did but it worked once... and I can't make it work again...

`
Sun Sep  4 06:40:17 2022: Starting logic: ParrotOverAudioLogic
Sun Sep  4 06:40:17 2022:       Found plugin: /usr/lib/arm-linux-gnueabihf/svxlink/RepeaterLogic.so
Sun Sep  4 06:40:17 2022: ParrotOverAudioLogic: Loading RX "HeadphoneInput"
Sun Sep  4 06:40:17 2022: ParrotOverAudioLogic: Loading TX "MicOutput"
Sun Sep  4 06:40:17 2022: ParrotOverAudioLogic: Loading module "ModuleParrot"
Sun Sep  4 06:40:17 2022:       Found /usr/lib/arm-linux-gnueabihf/svxlink/ModuleParrot.so
Sun Sep  4 06:40:17 2022:       Module Parrot v1.1.1 starting...
Sun Sep  4 06:40:17 2022: ParrotOverAudioLogic: Event handler script successfully loaded.
Sun Sep  4 06:40:26 2022: HeadphoneInput: The squelch is OPEN (513)
Sun Sep  4 06:40:26 2022: HeadphoneInput: The squelch is CLOSED (445)
Sun Sep  4 06:40:33 2022: HeadphoneInput: The squelch is OPEN (504)
Sun Sep  4 06:40:33 2022: HeadphoneInput: The squelch is CLOSED (422)
Sun Sep  4 06:41:10 2022:
`

I know it's a dumb request but can someone tell me what I'm doing wrong.

f5vmr, 2022-09-04:
From my perspective I see that CARD_CHANNELS=1 must be uncommented. Then
check with aplay -l what the usb sound card number is exactly. It will be
either 0 or 1 depending on whether you have blacklisted the on board sound
device. On the earlier models of the cm108 both poles of the microphone
socket were connected together, but the later models are true stereo
microphone inputs. If you are using the tips of the plugs then you should
use 0 or the band you use 1.
Finally to use the name you have selected for the logic, you need a
corresponding name in the /use/share/svxlink/events.d/local folder that not
only matches the name xxxxxx.tcl but the name of the logic to be edited
inside the first lines in the logic.

I have not read the log files you have displayed as they are not easily
readable in the form they appear. If you run the app from a service then
sudo system restart svxlink | tail -f /var/log/svxlink   (or .log) if you
modified the /etc/default/svxlink file.

Finally to initiate the logic you will still need to issue a carrier with
dtmf 1# unless you modify the Start up to do it. You can force the logic to
start up but I can't reference it here.

Chris
G4nab

WriteCodeEveryday, 2022-09-04:
Okay, I'll try the card channels, I just need this to parrot the input back to the headphones.

WriteCodeEveryday, 2022-09-04:
Yep, starts up successfully now

Sun Sep  4 20:35:24 2022: Starting logic: ParrotOverAudioLogic
Sun Sep  4 20:35:24 2022:       Found plugin: /usr/lib/arm-linux-gnueabihf/svxlink/RepeaterLogic.so
Sun Sep  4 20:35:24 2022: ParrotOverAudioLogic: Loading RX "MicInput"
Sun Sep  4 20:35:24 2022: ParrotOverAudioLogic: Loading TX "HeadphoneOutput"
Sun Sep  4 20:35:24 2022: ParrotOverAudioLogic: Loading module "ModuleParrot"
Sun Sep  4 20:35:24 2022:       Found /usr/lib/arm-linux-gnueabihf/svxlink/ModuleParrot.so
Sun Sep  4 20:35:24 2022:       Module Parrot v1.1.1 starting...
Sun Sep  4 20:35:24 2022: ParrotOverAudioLogic: Event handler script successfully loaded.

@f5vmr, is there a way to figure out what channel belongs to headphones or mic? I just want the mic input to be parroted back to the headphone input. My alsa device is correctly configured as the default ignoring the device default.

f5vmr, 2022-09-04:
You might want to consider creating a loopback dummy sound card. This will
feed back input to output directly. I use this function to echo output of a
repeater to a streaming web page using Darkice and Icecast.
Chris

WriteCodeEveryday, 2022-09-04:
@f5vmr Unfortunately that's not gonna work, I want Parrot after squelch is complete....

This is gonna be hooked up to HAM radio so I need the TX to occur after RX but overall this is doing exactly what I need... headphone output actually gives me my ident correctly, it's just that the mic input isn't played back over the headphone output.

WriteCodeEveryday, 2022-09-13:
`[ParrotOverAudioLogic]
TYPE=Simplex
RX=MicInput
TX=HeadphoneOutput
MODULES=ModuleParrot,ModuleHelp
CALLSIGN=KO4YZK
REPEATER_DELAY=1000
MUTE_RX_ON_TX=1
MUTE_TX_ON_RX=1
IDENT_ONLY_AFTER_TX=1
SHORT_IDENT_INTERVAL=1
EXEC_CMD_ON_SQL_CLOSE=200
EVENT_HANDLER=/usr/share/svxlink/events.tcl
DEFAULT_LANG=en_US
RGR_SOUND_DELAY=0
FX_GAIN_NORMAL=0
FX_GAIN_LOW=-12
IDLE_TIMEOUT=30
IDLE_SOUND_INTERVAL=3000
`

I rewrote my config and tried the DTMF tones for checking the help menu but it can't seem to work. I can key up and see the Squelch is open but I get no response and no return from my DTMF -> "*#"

Tue Sep 13 04:27:34 2022: MicInput: The squelch is OPEN (72)
Tue Sep 13 04:27:36 2022: MicInput: The squelch is CLOSED (5)

WriteCodeEveryday, 2022-09-13:
Looking at this again, I'm finding that I need +48db PREAMP just to get the VOX to open up on my radio for short ident.