sm0svx/svxlink #597: Multiple SvxReflector

gcm70, 2022-09-07:
Hi Tobias ,
have you planned or thought about the possibility of connecting multiple SvxReflector?
73

f5vmr, 2022-10-11:
Multiple reflectors are already in use, with differing results.
The rrf4.f5nlg.ovh:82 still on v1 protocol has a server on v2 and talk
groups. The svxportal-uk.ddns.net:81 has connections to the Swedish Portal.
I already have a node on two reflectors but with individual connection
between them.
Success is not the word I would use but it seems to work.
Chris.

dl1hrc, 2022-11-03:
In addition, I would like the possibility that the audio can be sent in duplex both between the nodes and the reflectors. Gladly also configurable
Mni 73s de Adi / DL1HRC

sm0svx, 2024-06-10:
My intention is to work on reflector interconnection code this summer. The first step will be to interconnect reflectors with the purpose of load sharing within a country. This interconnection variant is not meant to interconnect reflectors in other countries with each other though. That is the next step.

So interconnection is at the top of my TODO-list.

zochowsd, 2024-06-10:
Thanks for reply. I'm very excited - this is a super GREAT NEWS. 

We are observing a huge growth of FM modulation . Many people are tend to fallback to FM from Digital Voice space - what is good - but also create challange of scaling infrastucture. We have also some natural grow seeds from PL perspective - ex. USA / UK where local comunities of polish speaking operators are growing.  

IMHO small steps approach with this topic is really good. First thing first. It's really good that you have plan for it. crossing fingers for idea.

What we can also see there is trend of IPv6 adoption. Many of telco providers like ORANGE or T-MOBILE started to active default settings with IPv6 connectivity - but svxlink works only with IPv4. This is very popular problem with mobile based cennectivity hotspots - especially with summertime / vacations. I hope you will find some time with this topic also. Crossing fingers for your plans and execution. 

If I can suggest some implementation for federation protocol - maybe easer would be to start shaping communication ref-ref with speperated / dedicated ports and based the code on ref-node protocol - IMHO this part is really well stabilised. Reflectors are working without any problems. Many thanks for this implementation.

If there is anything that we can help with it - we are very happy to help.

f5vmr, 2024-06-10:
I concur… Small take up so far in the UK, but hopefully increasing.

Chris

dzlabing, 2025-03-05:
A possible starting point could be that a node uses different reflectors for different TGs. This approach would not require a dynamic routing protocol between reflectors and nodes, but might require an update to the node code. This update would allow for parallel connections to multiple reflectors, separated by TGs. 

For example, TGs 240, 2402, and 2403 could connect to a Swedish reflector, while TG 242 could connect to a Norwegian reflector.

sm0svx, 2025-03-16:
I am not aiming for nodes switching between reflectors. That would be waste of time and a suboptimal solution. The biggest problem really isn't the interconnection technology or protocol.

The fact that has stopped me most from progressing is how to authenticate reflector servers. And I don't mean in a technical sense but rather in a authenticity sense (does that make sense? :-D ). How do we verify that it really is a certified radio ham that want to connect their reflector server(s) to the network?  It can be a lot of work...

But I have decided to just continue with an implementation based on X.509 certificate technology now. Then we need to solve how certificates is to be issued, and withdrawn, in an acceptably safe way without too much work.