sm0svx/svxlink #598: Bug in Echolink Proxy Module?

DG1FI, 2022-09-12:
Hi!

I use two svxlink repeater behind a echolink proxy because no portforwading is possible. This works fine since many
yaers. Sometimes svxlink can't reach the proxy, in this condition svxlink crashes in not defined time
periodes and start again. No error message in the log.

Today i disabled the proxy for testing purposes, so svxlink can't reach them. And now it crashes again
and again . In the current version, the bug still exist.

73 Mario
DG1FI

sp2ong, 2022-10-03:
I confirm the problem with PROXY Echolink 

2022-10-03 09:41:05: *** ERROR: Command timeout while communicating to the directory server
2022-10-03 09:41:16: Disconnected from EchoLink proxy 172.67.161.142:8100
2022-10-03 09:41:37: SvxLink v1.7.99.69 Copyright (C) 2003-2022 Tobias Blomberg / SM0SVX
.....
2022-10-03 09:43:36: *** ERROR: Command timeout while communicating to the directory server
2022-10-03 09:43:47: Disconnected from EchoLink proxy 172.67.161.142:8100
2022-10-03 09:44:08: SvxLink v1.7.99.73 Copyright (C) 2003-2022 Tobias Blomberg / SM0SVX
2022-10-03 09:44:08: 
...
2022-10-03 09:47:06: *** ERROR: Command timeout while communicating to the directory server
2022-10-03 09:47:16: Disconnected from EchoLink proxy 172.67.161.142:8100
2022-10-03 09:47:37: SvxLink v1.7.99.73 Copyright (C) 2003-2022 Tobias Blomberg / SM0SVX
..
2022-10-03 09:49:37: *** ERROR: Command timeout while communicating to the directory server
2022-10-03 09:49:48: Disconnected from EchoLink proxy 104.21.57.82:8100
2022-10-03 09:50:08: SvxLink v1.7.99.73 Copyright (C) 2003-2022 Tobias Blomberg / SM0SVX
...
and svxlink crash again, again and again ....

about every 3 minutes svxlink restart and say a voice that reconnected to svxreflector ... 

73 Waldek

DG1FI, 2022-10-03:
Hi Waldek,

thank you very much for confirming the "proxy bug". Is there any start "verbose" or "debug" option, so i can put the output here.

73 Mario

DG1FI, 2022-10-23:
Today i rebooted Echolink Proxy Server, the svxlink client chrashed again.

dl1hrc, 2022-10-24:
I have a proxy running and it's doesn't happen here. Can you run svxlink with gdb to get more error/log info?
73s de Adi / DL1HRC

sp2ong, 2022-10-24:
Adi,

You can simulate this by writing in ModuleEchoLink.conf
proxy IP address that is unreachable and will be TIME OUT
For example, enter a dummy address in proxy_server ip that is not reachable, for example

PROXY_SERVER = 178.168.1.250
PROXY_PORT = 8100
PROXY_PASSWORD = PUBLIC

DG1FI, 2023-03-26:
Today i restart the echolink-proxy, svxlink crashed and restart again. The bug is still alive...

sm0svx, 2023-04-02:
Cannot reproduce this problem in latest git master. Make sure that you are using the latest master branch from my repository (sm0svx) when testing. Also ensure that no "contrib" code is enabled and that no thirdparty patches are applied.

draems, 2023-06-23:
Is this problem solved already?
I am experiencing the same problem. Port forwarding is not possible with my current service provider, so I installed a proxy application on an EC2 instance in AWS.

Fri Jun 23 09:32:36 2023: SimplexLogic: Event handler script successfully loaded.
Fri Jun 23 09:32:36 2023: Connected to EchoLink proxy X.X.X.X:8100
Fri Jun 23 09:32:37 2023: EchoLink directory status changed to ON
Fri Jun 23 09:32:38 2023: --- EchoLink directory server message: ---
Fri Jun 23 09:32:38 2023: EchoLink Server v2.6.118
Fri Jun 23 09:32:38 2023:
Fri Jun 23 09:32:38 2023: ECHO4: Frankfurt, Germany
Fri Jun 23 09:32:38 2023:
Fri Jun 23 09:39:36 2023: *** ERROR: Command timeout while communicating to the directory server

sp2ong, 2024-03-24:
**I confirm that still exists the problem with PROXY Echolink in SVXLink v.8**

2024-03-24 01:31:10: *** WARNING[getaddrinfo]: Could not look up host "elp187.pi9noz.ampr.org": Name or service not know
n
2024-03-24 01:31:10: Disconnected from EchoLink proxy 255.255.255.255:8100
2024-03-24 01:31:31: SvxLink v1.8.0@24.02-3-gcde00792 Copyright (C) 2003-2023 Tobias Blomberg / SM0SVX
............................
2024-03-24 01:31:41: *** WARNING[getaddrinfo]: Could not look up host "elp187.pi9noz.ampr.org": Name or service not know
n
2024-03-24 01:31:41: Disconnected from EchoLink proxy 255.255.255.255:8100
024-03-24 01:32:02: SvxLink v1.8.0@24.02-3-gcde00792 Copyright (C) 2003-2023 Tobias Blomberg / SM0SVX
2024-03-24 01:32:02: 
...................
2024-03-24 01:32:13: *** WARNING[getaddrinfo]: Could not look up host "elp187.pi9noz.ampr.org": Name or service not know
n
2024-03-24 01:32:13: Disconnected from EchoLink proxy 255.255.255.255:8100
2024-03-24 01:32:34: SvxLink v1.8.0@24.02-3-gcde00792 Copyright (C) 2003-2023 Tobias Blomberg / SM0SVX
2024-03-24 01:32:34: 
2024-03-24 01:32:34: SvxLink comes with ABSOLUTELY NO WARRANTY. This is free software, and you are