sm0svx/svxlink #600: Flatpak for the echolink client

joergklein, 2022-10-02:
Can any of you create a flatpak for Qtel? I myself have created a Docker container for svxlink. The downside is that Docker must also be installed on the notebook.
https://github.com/joergklein/docker-svxlink.git
https://hub.docker.com/r/joergklein/svxlink

f5vmr, 2022-10-02:
For me I have never used docker, and in any case I compile svxlink with sudo cmake -DUSE_QT=OFF -DCMAKE_INSTALL_PREFIX=/usr -DSYSCONF_INSTALL_DIR=/etc \ -DLOCAL_STATE_DIR=/var -DWITH_SYSTEMD=ON  ..

So QT is OFF.

QT doesn’t work in Debian11 (yet) This might not help you.

Chris

AsciiWolf, 2024-02-25:
I am working on a Qtel Flatpak. @sm0svx and Joerg, feel free to let me know whether you are interested in co-maintaining the Flatpak when it gets published on [Flathub](https://flathub.org). Thanks!

joergklein, 2024-02-26:
Hello AsciiWolf

Thank you for your Mail. I Think it is a great idea to have Qtel on
Flatpak. I have no idea how to program for Flatpak, but I will learn it.

Here is my Docker Container for  svxlink. Please take a look at
https://github.com/joergklein/docker-svxlink

Thanks
Jörg

AsciiWolf, 2024-02-26:
Hello Jörg,

Feel free to let us know in [the ticket](https://github.com/flathub/flathub/pull/4995) whether you are interested in co-maintaining (having push rights to its Git repository) the Flatpak on Flathub.

Thanks! :-)

sm3sgp, 2025-03-18:
@joergklein @AsciiWolf Looks like this issue was fixed?

closed 2025-03-18 by sm3sgp

AsciiWolf, 2025-03-18:
It was. The only problem is that the Flatpak is not [verified](https://docs.flathub.org/docs/for-users/verification) because upstream is not involved in its maintenance. This makes the Flatpak hidden by default on some distros such as Linux Mint. But it works fine and already has over 3000 users.