sm0svx/svxlink #601: ERROR: Unable to handle event - (invalid command name "SimplexLogic_70cm::every_second")

tsk7mde, 2022-10-06:
Hello,

I get the following error as soon as I rename the logic, it works, but the error comes up constantly, what can I do. 
Can someone support me? 

*** ERROR: Unable to handle event: SimplexLogic_70cm::every_second in logic SimplexLogic_70cm (invalid command name "SimplexLogic_70cm::every_second")

Thanks and regards

**_Log:_**

SvxLink v1.7.99.73 Copyright (C) 2003-2022 Tobias Blomberg / SM0SVX

SvxLink comes with ABSOLUTELY NO WARRANTY. This is free software, and you are
welcome to redistribute it in accordance with the terms and conditions in the
GNU GPL (General Public License) version 2 or later.

Using configuration file: /etc/svxlink/svxlink.conf
--- Using sample rate 48000Hz

Starting logic: SimplexLogic_70cm
        Found plugin: /usr/lib64/svxlink/SimplexLogic.so
SimplexLogic_70cm: Loading RX "Rx1_70cm"
SimplexLogic_70cm: Loading TX "Tx1_70cm"
SimplexLogic_70cm: Loading module "ModuleHelp"
        Found /usr/lib64/svxlink/ModuleHelp.so
        Module Help v1.0.0 starting...
SimplexLogic_70cm: Loading module "ModuleParrot"
        Found /usr/lib64/svxlink/ModuleParrot.so
        Module Parrot v1.1.1 starting...
SimplexLogic_70cm: Event handler script successfully loaded.
*** ERROR: Unable to handle event: SimplexLogic_70cm::startup in logic SimplexLogic_70cm (invalid command name "SimplexLogic_70cm::startup")
*** ERROR: Unable to handle event: SimplexLogic_70cm::every_second in logic SimplexLogic_70cm (invalid command name "SimplexLogic_70cm::every_second")
*** ERROR: Unable to handle event: SimplexLogic_70cm::every_second in logic SimplexLogic_70cm (invalid command name "SimplexLogic_70cm::every_second")
*** ERROR: Unable to handle event: SimplexLogic_70cm::every_second in logic SimplexLogic_70cm (invalid command name "SimplexLogic_70cm::every_second")
*** ERROR: Unable to handle event: SimplexLogic_70cm::every_second in logic SimplexLogic_70cm (invalid command name "SimplexLogic_70cm::every_second")
*** ERROR: Unable to handle event: SimplexLogic_70cm::every_second in logic SimplexLogic_70cm (invalid command name "SimplexLogic_70cm::every_second")

dl1hrc, 2022-10-06:
Hi,
you have to create a namespace **SimplexLogic_70cm** in the tcl-environment too. The easiest way is to copy the /usr/share/svxlink/events.d/SimplexLogic.tcl to /usr/share/svxlink/events.d/local/SimplexLogic_70cm.tcl and change the entry at the beginning of this file like this:

```
#
# This is the namespace in which all functions below will exist. The name
# must match the corresponding section "[SimplexLogic]" in the configuration
# file. The name may be changed but it must be changed in both places.
#
namespace eval SimplexLogic_70cm {
....
```

Mni 73 de Adi / DL1HRC

tsk7mde, 2022-10-06:
Thanks very much! That solved my problem.
Greeting

closed 2022-10-06 by tsk7mde