sm0svx/svxlink #614: Problem with HOSTS handling in ReflectorLogic

dl1hrc, 2023-03-05:
I think there is a problem with the handling of HOSTS in ReflectorLogic:
```
  std::vector<std::string> hosts;
  if (cfg().getValue(name(), "HOST", hosts))
  {
  ...
  }
  cfg().getValue(name(), "HOSTS", hosts);
```

From time to time the assignment of the HOSTS entries to the vector elements is failing and the string is split at the decimals. So the Logic can not connect to the server since it tries to open a connection to
"abc"
"def"
"com"
if the configured hostname is "abc.def.com"
It can be reproduced here but it doesn't happen everytime when ste svxlink server starts.
Here is a possible solution:

`````
diff --git a/src/svxlink/svxlink/ReflectorLogic.cpp b/src/svxlink/svxlink/ReflectorLogic.cpp
index abea891d..fdc608e2 100644
--- a/src/svxlink/svxlink/ReflectorLogic.cpp
+++ b/src/svxlink/svxlink/ReflectorLogic.cpp
@@ -49,6 +49,7 @@ Foundation, Inc., 59 Temple Place, Suite 330, Boston, MA  02111-1307  USA
 #include <AsyncAudioPassthrough.h>
 #include <AsyncAudioValve.h>
 #include <version/SVXLINK.h>
+#include <common.h>
 
 
 /****************************************************************************
@@ -197,20 +198,26 @@ bool ReflectorLogic::initialize(Async::Config& cfgobj, const std::string& logic_
   cfg().getValue(name(), "VERBOSE", m_verbose);
 
   std::vector<std::string> hosts;
-  if (cfg().getValue(name(), "HOST", hosts))
+  std::string myhost;
+  if (cfg().getValue(name(), "HOST", myhost))
   {
     std::cout << "*** WARNING: The " << name()
               << "/HOST configuration variable is deprecated. "
                  "Use HOSTS instead." << std::endl;
+    hosts.push_back(myhost);
+  }
+  else
+  {
+    cfg().getValue(name(), "HOSTS", myhost);
+    SvxLink::splitStr(hosts, myhost, ",");
   }
-  cfg().getValue(name(), "HOSTS", hosts);
   std::string srv_domain;
   cfg().getValue(name(), "DNS_DOMAIN", srv_domain);
   if (srv_domain.empty() && hosts.empty())
   {
     std::cerr << "*** ERROR: At least one of HOSTS or DNS_DOMAIN must be "
                  "specified in " << name() << std::endl;
-     return false;
+    return false;
   }
 
   if (!srv_domain.empty())

sm0svx, 2023-03-07:
I cannot reproduce this problem. Which commit ID did you test on?

dl1hrc, 2023-03-08:
just tested with last commit:
`SvxLink v1.7.99.77 Copyright (C) 2003-2023 Tobias Blomberg / SM0SVX`

System:
`Raspbian GNU/Linux 11 \n \l`
`Linux DO0KPB-70cm-FM 5.15.84-v7+ #1613 SMP Thu Jan 5 11:59:48 GMT 2023 armv7l GNU/Linux`

Startlog:
```
Using configuration file: /etc/svxlink/svxlink.conf
--- Using sample rate 48000Hz

Starting logic: RepeaterLogic
	Found plugin: /usr/lib/arm-linux-gnueabihf/svxlink/RepeaterLogic.so
RepeaterLogic: Loading RX "Rx1"
Starting ZVEI1 decoder
RepeaterLogic: Loading TX "Tx1"
RepeaterLogic: Loading module "ModuleHelp"
	Found /usr/lib/arm-linux-gnueabihf/svxlink/ModuleHelp.so
	Module Help v1.0.0 starting...
RepeaterLogic: Loading module "ModuleParrot"
	Found /usr/lib/arm-linux-gnueabihf/svxlink/ModuleParrot.so
	Module Parrot v1.1.1 starting...
RepeaterLogic: Loading module "ModuleEchoLink"
	Found /usr/lib/arm-linux-gnueabihf/svxlink/ModuleEchoLink.so
	Module EchoLink v1.5.99.3 starting...
*** ERROR: Please set the EchoLink callsign (ModuleEchoLink/CALLSIGN) to a real callsign
*** ERROR: Initialization failed for module ModuleEchoLink in logic RepeaterLogic
RepeaterLogic: Loading module "ModuleTclVoiceMail"
	Found /usr/lib/arm-linux-gnueabihf/svxlink/ModuleTcl.so
	Module Tcl v1.0.1 starting...
RepeaterLogic: Loading module "ModuleMetarInfo"
	Found /usr/lib/arm-linux-gnueabihf/svxlink/ModuleMetarInfo.so
	Module MetarInfo v1.2.1.99.4 starting...
RepeaterLogic: Event handler script successfully loaded.

Starting logic: SipLogic
	Found plugin: /usr/lib/arm-linux-gnueabihf/svxlink/SipLogic.so
SipLogic: Simplexmode, using VOX squelch for Sip.
SipLogic: Event handler script successfully loaded.
SipLogic: Event handler script successfully loaded.
>>> Started SvxLink with special SipLogic extension (v27112022)
>>> No guarantee! Please send a bug report to
>>> Adi/DL1HRC <dl1hrc@gmx.de> or use the groups.io mailing list

Starting logic: ReflectorLogicKWD
	Found plugin: /usr/lib/arm-linux-gnueabihf/svxlink/ReflectorLogic.so
ReflectorLogicKWD: Connecting to service 

Starting logic: ReflectorLogicTHR
	Found plugin: /usr/lib/arm-linux-gnueabihf/svxlink/ReflectorLogic.so
ReflectorLogicTHR: Connecting to service 
Activating link NetLinkTHR
Activating link SipLink
*** WARNING[getaddrinfo]: Could not look up host "svxlink": Name or service not known
ReflectorLogicKWD: Disconnected from 255.255.255.255:xxxx: Host not found
*** WARNING[getaddrinfo]: Could not look up host "master1": Name or service not known
ReflectorLogicTHR: Disconnected from 255.255.255.255:yyyy Host not found
*** WARNING[getaddrinfo]: Could not look up host "ham-radio-op": Name or service not known
ReflectorLogicKWD: Disconnected from 255.255.255.255:xxxx: Host not found
*** WARNING[getaddrinfo]: Could not look up host "thueringen": Name or service not known
ReflectorLogicTHR: Disconnected from 255.255.255.255:yyyy: Host not found
*** WARNING[getaddrinfo]: Could not look up host "net": Name or service not known
ReflectorLogicKWD: Disconnected from 255.255.255.255:54: Host not found
*** WARNING[getaddrinfo]: Could not look up host "link": Name or service not known
ReflectorLogicTHR: Disconnected from 255.255.255.255:xxxx: Host not found
Connected to APRS server 82.223.33.109 on port 14580
```
svxlink.conf:
```
# Link zum Türingen Link
[ReflectorLogicTHR]
TYPE=Reflector
HOSTS=master1.thueringen.link
HOST_PORT=xxxxx
CALLSIGN="DO0xxxxx"
AUTH_KEY="xxxxxx"
JITTER_BUFFER_DELAY=100
DEFAULT_TG=26207
MONITOR_TGS=1,945,26207,263,14774,26221
TG_SELECT_TIMEOUT=60
ANNOUNCE_REMOTE_MIN_INTERVAL=50
MUTE_FIRST_TX_LOC=0
MUTE_FIRST_TX_REM=0
EVENT_HANDLER=/usr/share/svxlink/events.tcl
NODE_INFO_FILE=/etc/svxlink/node_info.json

[ReflectorLogicKWD]
TYPE=Reflector
HOSTS=svxlink.ham-radio-op.net:xxxx
HOST_PORT=xxxxx
CALLSIGN="DOxxx-1"
AUTH_KEY="xxxxxxxxx"
JITTER_BUFFER_DELAY=300
DEFAULT_TG=26208
MONITOR_TGS=9,2629,26208,9111,9112
ANNOUNCE_REMOTE_MIN_INTERVAL=300
EVENT_HANDLER=/usr/share/svxlink/events.tcl
NODE_INFO_FILE=/etc/svxlink/node_info.json
MUTE_FIRST_TX_LOC=0
MUTE_FIRST_TX_REM=0
QSY_PENDING_TIMEOUT=15
```

It seems that the parameters of both reflector sections are "mixed".

sm0svx, 2023-03-08:
ok I have never tested using two reflectors at the same time. Do you get the same problem if you remove one of the reflector configs?  What if you just keep the RepeaterLogic and one ReflectorLogic (no SipLogic)?

f5vmr, 2023-03-08:
I too have been round svxlink for some time, and have enjoyed the whole concept. As for using two Reflectors at the same time, I have. 

I just use two ReflectorLogic, two reflector Logic links, and use ‘9’ for one and ‘8’ for the other, suppressing the the QSO recorder as I have been using ‘8’.

So I was using 81+ TG for one network, and 91+TG for the other. 

The UK reflector is not well populated, and the French reflector to which I connected has had the link to the French RRF removed. Jean-Phillipe is reluctant to get his clientele to change to the Protocol 2 of the Reflector Logic…. I pity as it is far superior to the way he was using the other protocol.

Keep up the good work.

Chris G4NAB/F5VMR that was.

dl1hrc, 2023-03-09:
I tested this:

1) one ReflectorLogic without SipLogic: works
2) one ReflectorLogic with SipLogic: works
3) two ReflectorLogics with SipLogic: fails in ~50% with the error messages above

I feel it hasn't occurred in the past (>5 month) but I don't know the exact time. It could also be a platform dependent issue as I haven't noticed it on other systems.
Could you please briefly explain how the allocation of the parameter "HOSTS" to the "std::vector hosts" in the ReflectorLogic class works? I didn't find a method that splits the comma separated string to vector elements in AsyncConfig class.

mni 73s de Adi / DL1HRC

sm0svx, 2023-04-02:
Adi, it's not obvious what function is used to split the config string but it's an [overloaded template function](https://github.com/sm0svx/svxlink/blob/master/src/async/core/AsyncConfig.h#L235-L288) that is called due to supplying the getValue function with a std::vector config variable.

Did you also test the case with two ReflectorLogics but without SipLogic?