sm0svx/svxlink #625: Add support for G.711 codec

dscp46, 2023-05-05:
Hi,

I'm operating with fellow OMs a regional FM repeater network with 5.7GHz links (Hamnet) as a backhaul. 
We're planning to open a reflector that will be designed to transport Data or SSTV over Narrowband FM.
The catch is we're currently stuck with a bandwidth issue because the RAW codec is quite hungry, and some of our links are not fast enough to reliably send those signals. 

Considering that G.711 easily covers a FM repeater's service bandwidth, and that it is well known in the telecom industry to be reliable to transport complex digital signals, would it be possible to implement it as a codec in svxlink and svxreflector to satisfy our needs?

I know there are a few challenges to address with simplex links, but they're not part of our scope, for the time being.

73 de F4HOF

f5nlg, 2023-05-05:
with opus codec it works on sstv here i had a sstv parrot with svxlink and it works like a charme .73"Jean-Philippe Piers - F5NLG

sm0svx, 2023-07-18:
Don't use the RAW codec. Its 512kbps (32 bit float 16ksps). If you want uncompressed audio, use the S16 codec instead which is 256kbps (16 bit signed integer 16ksps). The Opus codec is very adaptable so if it does not work with the standard settings you can try to tweak some parameters. The easiest one to try first, on all nodes sending digital signals to the reflector, is to set OPUS_ENC_BITRATE=64000. That will give you the same bitrate as G.711. The default is 20000.

Note though that the SvxLink application is not written with digital signals in mind so some digital filters that are applied may distort digital signals.

closed 2025-03-18 by sm3sgp